peterhinch/micropython-async #1: How to make uasyncio.core a frozen module

mkarliner, 2017-01-28:
I'm probably being incredibly dim, but I'm having great difficulty including uasyncio.core a frozen module on the esp8266. It needs to be frozen, as it's too big to upip or import from the file system. I'm symbolically linking from the micropython-lib directory.  
First, the dot in the directory name causes compile errors on the build thusly:
build/frozen_mpy.c:4121:41: error: expected '=', ',', ';', 'asm' or '__attribute__' before '.' token
 STATIC const byte bytecode_data_uasyncio.core_example_call_soon__lt_module_gt__cb[52] = {

Second, the uasyncio.core module has no __init__.py file, so I don't see how it can be used as a module as is.

I'm asking this here, as other aspirant pythonistas may be similarly stuck and this seemed a good place. Apologies if this is not a good place. Could you give me any pointers on the correct way to do this?
Thanks in advance.

peterhinch, 2017-01-29:
I haven't actually tested this with the ESP8266 but here is how I've done it on the Pyboard. Create the Unix build on your PC and install the library as per the instructions [here](https://github.com/micropython/micropython-lib). This will create a directory ~/.micropython/lib which includes the necessary dependencies and __init__.py

Then copy or symlink lib into your modules directory and compile.

craftyguy, 2017-03-06:
Maybe I'm way more dim than the submitter of this issue, but I seem to be lost as to what I should be doing with the uasyncio module (which is now installed under ~/.micropython/lib per your instructions). 

I have a simple python program that I would like to use with uasyncio, but it's unclear how I now build a version of this specific to the esp8266. Is the best way to somehow integrate this module into micropython so it's included when I build the esp8266-specific micropython image from source?

craftyguy, 2017-03-06:
I found this excellent article from adafruit that walked me through the process: https://learn.adafruit.com/micropython-basics-loading-modules/frozen-modules

Can't wait to get home and see if it was successful! (and sorry for the ticket spam, hopefully this helps others like me!)

mkarliner, 2017-03-06:
I'm the OP in question, and I'm still dim. It took me a while work this one out, and in the end it wasn't that difficult. 
What I did was make an uasyncio directory in my modules directory and then put core.py from uasyncio.core  and __init__.py from uasyncio  into it. After a build and deploy I can now do an import uasyncio without problems. 

I will change this procedure to do a symbolic link instead of a copy, and actually spend some time trying out uasyncio, as all I've had time for is typing 'import', but hopefully this will help you and those that follow.

closed 2017-04-21 by peterhinch

chrisovergaauw, 2020-03-18:
Hi,

My apologies for necroing this issue, but upon goolging "how to freeze uasyncio"
this is the only answer which comes close to providing an explanation and in my opinion it is incomplete. (I came to this conclusion thanks to my own dim-wittedness)

As mentioned by  craftguy the adafruit article perfectly covers how to freeze custom modules and mkarliner sheds light on the actual directory structure required for uasyncio, but what is missing is how one would obtain the files.

I'm guessing one way would be to go through the source code and copy each respective file but it is hard to imagine that there's no better alternative out there.
I've tried looking for solutions to install upip on OSX and because doing regular pip installs for the micropython libraries always results in errors for me.

 Would anyone care to enlighten me?

regards,
Chris

peterhinch, 2020-03-19:
There are two easy ways to get the files onto your PC. You can run upip under the Unix build of MicroPython. If you haven't got the Unix build, I wrote micropip [in this repo](https://github.com/peterhinch/micropython-samples) which does the same job under Python 3. It has been tested on OSX.

The less easy way is to clone [micropython-lib](https://github.com/micropython/micropython-lib) using
```bash
git clone https://github.com/micropython/micropython-lib
```
and copy the relevant files.

chrisovergaauw, 2020-03-19:
Thank you. Using your tool was indeed faster and easier than copying files manually.