peterhinch/micropython-async #4: uasyncio open_connection seems broken after freezing it

riegaz, 2017-05-14:
I just started a issue there:

https://github.com/micropython/micropython-lib/issues/177

for me after freezing uasyncio it seems that yield from open_connection() never returns.

I'm investigating...

riegaz, 2017-05-14:
I just added an output within the function:

```
    def call_later_ms_(self, delay, callback, args=()):
	print('delay '+str(delay))
        self.call_at_(time.ticks_add(self.time(), delay), callback, args)
```
And it shows me 1045220558000 as delay. so somewhere the delay variable becomes wrong.

peterhinch, 2017-05-15:
Thank you for pointing this out. Given that this occurs with the official build this is an issue for the maintainers but I'll monitor #177 and ensure my fork implements any changes.

riegaz, 2017-05-17:
Solved https://github.com/micropython/micropython/commit/88c51c3592edc6c29fb0aad57c91e535793aa31b

closed 2017-05-17 by riegaz