peterhinch/micropython-async #5: I'm trying to use aremote.py togother with picoweb, and thing works werid

fruch, 2017-07-24:
I'm doing something like this:
```python
ROUTES = [
    # You can specify exact URI string matches...
    ("/", lambda req, resp: (yield from app.sendfile(resp, "index.html"))),
    ("/favicon.ico", lambda req, resp: (yield from app.sendfile(resp, "favicon.ico"))),
    ("/tinycolorpicker.css", lambda req, resp: (yield from app.sendfile(resp, "tinycolorpicker.css"))),
    ("/tinycolorpicker.js", lambda req, resp: (yield from app.sendfile(resp, "tinycolorpicker.js"))),
    ("/text-color.png", lambda req, resp: (yield from app.sendfile(resp, "text-color.png"))),
    ("/rgb/", rgb),
]
            
def cb(data, addr):
    if data == REPEAT:
        print('Repeat')
    elif data >= 0:
        print(hex(data), hex(addr))
    else:
        print('{} Address: {}'.format(data, hex(addr)))

print('Test for IR receiver. Assumes NEC protocol.')
machine.freq(160000000)
p = machine.Pin(14, machine.Pin.IN)
ir = NEC_IR(p, cb, True)  # Assume r/c uses extended addressing
    
app = picoweb.WebApp(__name__, ROUTES)
app.run(debug=False, host="0.0.0.0")
```

and aremote seem to be working fine, but every http request is blocked until pressing a remote key.
couldn't figure out what fishy here...

peterhinch, 2017-07-25:
That's got me beat, too. I haven't used picoweb but from a quick look at the code they should play together. While waiting for a signal from the remote, at line 57 in aremote.py, it waits on the Event instance's `__iter__` / `__await__` method which waits on `sleep(0)`. So other coroutines should be scheduled in round-robin fashion. You could try putting debug print statements there to verify that's what's happening.

I'd also be tempted as a test to omit the IR stuff and write a simple coro which does something like periodically print "hello" to prove picoweb behaves sociably.

It may be a few days before I get a chance to look at this more closely.

peterhinch, 2017-07-25:
See https://github.com/micropython/micropython/issues/3242

closed 2018-11-30 by peterhinch