peterhinch/micropython-async #7: Suggestion: rename CancelError

antocuni, 2017-12-11:
This is more a suggestion than an issue, but since I saw that the commits about `NamedCoro` etc. are very recent, I thought it might be a good time to change it before it's too late :).

I think that `CancelError` is very confusing: 
- it's not really an error, it's a part of the flow control: in the same way, we use `StopIteration` instead of `StopIterationError`

- at a first look, it's unclear whether it means that the task was canceled, or whether there was an error while canceling it

Some proposals for a better name:

- `Canceled`
- `NamedCoro.Stop` or `NamedCoro.Canceled`
- `AsyncStop`
- `StopCoroutine`, which is nice because it's similar to `StopIteration`

peterhinch, 2017-12-12:
Thank you. I agree, I think `StopCoroutine` is perhaps best for the reason you give; although I see Paul is also using the term `task`.

So I wonder if `NamedCoro` should become `NamedTask` and the exception should be `StopTask`? After all `NamedCoro` already has a `task` bound variable.

What do you think?

antocuni, 2017-12-12:
I like `NamedTask`, especially considering that you start it using `loop.create_task`. 
Then to avoid confusion, the current `NamedCoro.tasks` needs probably to be renamed.

I propose the following:
```python

class NamedTask:
    class Stop(Exception):
        pass
    ...
```

cefn, 2018-01-03:
Is this equivalent to the CPython name CancelledError ?

https://docs.python.org/3/library/concurrent.futures.html?highlight=cancellederror#concurrent.futures.CancelledError

In CPython, Task#cancel() raises CancelledError within the coroutine itself (at the await) in the same way as calling coro.throw(CancelledError).

https://docs.python.org/3/library/asyncio-task.html?highlight=task%20cancel#asyncio.Task.cancel

peterhinch, 2018-01-04:
In uasyncio task cancellation works rather differently compared to CPython asyncio. I've attempted to explain this [here](https://github.com/peterhinch/micropython-async/blob/master/TUTORIAL.md#36-task-cancellation). In essence **pend_throw()** is used to throw an exception to the task. This will be handled when the task is next scheduled. This is the mechanism used to implement timeouts.

closed 2018-11-30 by peterhinch