peterhinch/micropython-async #8: DGRAM support?

ranma1988, 2017-12-31:
Im trying to use socket.recvfrom for DNS server please add DGRAM support.

peterhinch, 2018-01-01:
This request should be targeted at the [MicroPython(https://github.com/micropython/micropython.git) repository. I am not responsible for the **socket** module.

closed 2018-01-01 by peterhinch

ranma1988, 2018-01-01:
Hi Peter,
the socket in MicroPython supports **recvfrom**.
My issue looks like this:

```
async def `dns(s):`
    while True:
      try:
        data, addr = yield s.recvfrom(1024)

loop = asyncio.get_event_loop()
loop.run_until_complete(dns2(udps))
loop.run_forever()
loop.close()
```

AssertionError: Unsupported coroutine yield value: (b'\x97w\x01\x00\x00\x01\x00\x00\x00\x00\x00\x00\x11connectivitycheck\x07gstatic\x03com\x00\x00\x01\x00\x01', ('192.168.4.2', 54332)) (of type <class 'tuple'>)

peterhinch, 2018-01-02:
There are two issues here. The first is DGRAM support which I believe is ongoing: queries should be addressed to the **socket** module maintainers. The second is your use of the **yield** keyword. The **recvfrom** method is synchronous so I'd expect it to be called with

```python
    data, addr = s.recvfrom(1024)
```

But queries on coding should be raised in [the forum](https://forum.micropython.org/) where they will reach a larger helpful audience.

GitHub issues are for bug tracking or feature requests by RFC. As far as I can see you have identified no bugs.