peterhinch/micropython-async #9: A more ample introduction is required for the complete beginner.

stroobandt, 2018-01-22:
With all due respect, Peter, but as a beginner I found it hard to jump head first into your otherwise excellent text.

Reading the web article [_Asynchronous Python for the Complete Beginner_](https://py.checkio.org/blog/asynchronous-python-for-the-complete-beginner/) first, helped with rendering your text more accessible.

So I would suggest linking or adding some of these elements to the introduction of your text.

Thanks!
Serge

peterhinch, 2018-01-23:
Thank you for your comments. I take your point that my "Notes for beginners" section 7 would benefit from a better introduction with simple initial code samples. I think I was answering the question "Why use `uasyncio` rather than `_thread`?" rather than "How do I begin?".

I'm not entirely happy with the web article since it is focussed on PC applications. It also strays off into discussions of multiple processors, the GIL, greenlets, old `@asyncio.coroutine` syntax and other issues likely to confuse the beginner wishing to program the Pyboard.

So I'll endeavour to improve section 7 while keeping the focus on MicroPython.

stroobandt, 2018-01-23:
> I'm not entirely happy with the web article since it is focussed on PC applications. It also strays off into discussions of multiple processors, the GIL, greenlets, old `@asyncio.coroutine` syntax and other issues likely to confuse the beginner wishing to program the Pyboard.

Yes, that passage is indeed completely superfluous, nor do I like the article's conclusion.
Nonetheless, that article helped me come to grabs with `asyncio`, whereas after reading your section 7, I was still baffled.

Beginners learning `asyncio` are confronted basically with two problems:

1. There are no textbooks available about the subject,
2. The module `asyncio` introduces a lot of non-evident syntactic vocabulary. A non-native English speaker may get quite confused with, for example, syntactic word combinations like `yield from`.

peterhinch, 2018-01-23:
Well I can't fix Python syntax (`yield from` is in the core language). But I will make an effort to improve section 7 :)

stroobandt, 2018-01-23:
I truly appreciate the efforts you make for the MicroPython community. You surely understand that I am trying to return some constructive feedback along the way of my learning process.

peterhinch, 2018-01-24:
Thank you. I do appreciate the feedback.

closed 2018-11-30 by peterhinch