peterhinch/micropython-async #11: First Asyncio example does not work as described.

mkarliner, 2018-01-25:
In the tutorial 
2.1 Program structure the event loop.
in  following code
```
import uasyncio as asyncio
loop = asyncio.get_event_loop()
async def bar():
    count = 0
    while True:
        count += 1
        print(count)
        await asyncio.sleep(1)  # Pause 1s

loop.create_task(bar()) # Schedule ASAP
loop.run_forever()
```

I note that loop is a PollEventLoop object, not an EventLoop object and thus has no create_task or run_forever methods. I assume I'm doing something wrong, but what?

peterhinch, 2018-01-25:
It does work on a Pyboard. The `PollEventLoop` class is a subclass of `EventLoop` and hence does have those methods.

So the issue is either with the installation of the library or with the Loboris port (I have used `uasyncio` on ESP32 but only with the official port).

What error message do you see? If it indicates that those methods are absent I suspect that the library may not be correctly installed. You might like to try this (done on a Pyboard):
```python
>>> import uasyncio
>>> dir(uasyncio)
['IOWrite', 'log', 'select', 'get_event_loop', 'IORead', 'Task', 'ensure_future', 'PollEventLoop', 'IOReadDone', 'StreamReader', 'uerrno', 'start_server', 'EventLoop', 'StopLoop', 'StreamWriter', 'uasyncio', 'time', 'cancel', 'sleep', 'utimeq', 'sleep_ms', 'CancelledError', 'SysCall1', 'wait_for_ms', 'TimeoutError', '_socket', 'SleepMs', 'set_debug', 'SysCall', 'coroutine', 'DEBUG', 'open_connection', 'IOWriteDone', 'type_gen', 'core', '__name__', 'wait_for', 'TimeoutObj', '__path__']
>>> 
```

mkarliner, 2018-01-25:
>>> import uasyncio
>>> dir(uasyncio)
['log', 'get_event_loop', 'select', 'Task', 'IORead', 'PollEventLoop', 'ensure_future', 'StreamReader', 'IOReadDone', 'uerrno', 'open_connection', 'start_server', 'EventLoop', 'StopLoop', 'StreamWriter', 'uasyncio', 'time', 'cancel', 'sleep', 'utimeq', 'sleep_ms', 'CancelledError', '__file__', 'SysCall1', 'wait_for_ms', 'TimeoutError', '_socket', 'SleepMs', 'set_debug', 'SysCall', 'coroutine', 'DEBUG', '__name__', 'IOWriteDone', 'type_gen', 'core', '__path__', 'wait_for', 'TimeoutObj', 'IOWrite']
>>>

mkarliner, 2018-01-25:
I note that PollEventLoop and EventLoop are separate.
I

mkarliner, 2018-01-25:
I've rebuilt from scratch an official image, and it does the same thing.

peterhinch, 2018-01-26:
Your `dir()` is the same as mine except you have an additional `__file__` entry which is because my `uasyncio` on the Pyboard is frozen as bytecode.

I've now installed the latest official firmware build on ESP32.

```
MicroPython v1.9.3-240-ga275cb0f on 2018-01-26; ESP32 module with ESP32
```
I deleted my old `uasyncio` installation and verified that it couldn't be imported. After connecting to WiFi I ran:

```python
import upip
upip.install('micropython-uasyncio')
upip.install('micropython-uasyncio.queues')
upip.install('micropython-uasyncio.synchro')
```

The test script now runs. What error message do you see when you run it?

My installation of `uasyncio` on ESP32 is in `/pyboard/lib/uasyncio` and contains `__init__.py core.py     queues.py   synchro.py`.

mkarliner, 2018-01-26:
I've followed your steps and can confirm that this works. However, the lib  directory is under the root and not pyboard. Is this a typo?
`>>> import os
>>> os.listdir()
['boot.py', 'lib', 'asynctest.py', 'aold.py']
>>> os.listdir('lib')
['uasyncio', 'collections']
>>> os.listdir('lib/uasycio')
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
OSError: [Errno 2] ENOENT
>>> os.listdir('lib/uasyncio')
['__init__.py', 'core.py', 'queues.py', 'synchro.py']
>>>`

The difference from the previous is only that I was running from a src build which was from the ESP32 port section of the main repo. The uasyncio install was in both cases via upip.

While I'm here, which is the current official source for ESP32, the main report or micropython-ESP32?

mkarliner, 2018-01-26:
After going through the whole cycle once more and reloading again from my from source image, I now have the example running as described. 
I'll put this down to the Lobo build and possibly take it up with him.
Thanks for the help

closed 2018-01-26 by mkarliner

peterhinch, 2018-01-26:
To answer your questions the directories appear under `/pyboard` if you're using `rshell`. From the point of view of MicroPython on the board it appears under `/`.

As for official source it's now in the main repo. I guess the old repo exists to preserve issues etc.

@Loboris seems ambivalent about `uasyncio`. I made some effort to convince him of its merits relative to threads but I'm not sure if I succeeded. The more people who pitch in the better - it is in everyone’s interest for his port properly to support `uasyncio`.

mkarliner, 2018-01-26:
Completely agree.