peterhinch/micropython-async #12: asyncio with multiple methods in microPython

Sunrise17, 2018-01-31:
Dear @peterhinch 
When i am executing following code, "say" methods and led toggle method, are executed asynchronously so well. However, when i add time.sleep(10) in "async def say ()" method, led stops toggling, until completion of while True in "async def say()" method. Is it possible in micropython, to execute both functions asynchronously before waiting the completion of while True in one of each? Because led toggle should never been interrupted in according to the execution of another method.

Thanks for your support,

////
import uasyncio as asyncio
from machine import Pin
import time

LED_PIN = 13
led = Pin(LED_PIN, Pin.OUT, value=1)

async def toggle():
    while True:
        await asyncio.sleep_ms(500)
        led_toggle()

def led_toggle():
    led.value(not led.value())

async def say(what, when):
    # time.sleep(10)  ## !! ((if time sleep added for delay, led stops toggling)) !!
    while True:
        await asyncio.sleep(when)
        print(what)

loop = asyncio.get_event_loop()
loop.create_task(toggle())
loop.create_task(say('hello', 0.5)) # Schedule ASAP
loop.create_task(say('world', 1)) # Schedule ASAP
loop.run_forever()
////

peterhinch, 2018-02-01:
I suggest you read my tutorial which explains why you should use `await asyncio.sleep(10)` and not `time.sleep(10)`.

Sunrise17, 2018-02-01:
Dear @peterhinch , thanks for your quick response, i am not familiar to work with asyncio, first of all sorry for my basic issues. May be i could not tell my issue good enough. To avoid any misunderstanding, i want to give another example as shown below (without any time.sleep or await asyncio.sleep) Because in that example i especially tried to create a virtual hang for that method with time.sleep. 

When i run the following code, it runs and print ("Listening, connect your APP to http://192.168.4.1:8080/") and waiting request as web server. During this period of time, LED should go on blinking but it stops here. Unless it receives any request (which activates WhileTrue loop in webServer), LED does not respond. How can i add async property to "while true loop" in "async def webServer(ipAddress)" method?

Thanks again for your support,
Kind regards...

```
import uasyncio as asyncio
from machine import Pin
import time

LED_PIN = 13
led = Pin(LED_PIN, Pin.OUT, value=1)

async def toggle():
    while True:
        await asyncio.sleep_ms(500)
        led_toggle()

def led_toggle():
    led.value(not led.value())

async def webServer(ipAddress):
    s = socket.socket()
    ai = socket.getaddrinfo(ipAddress, 8080)
    print("Bind address info:", ai)
    addr = ai[0][-1]
    s.setsockopt(socket.SOL_SOCKET, socket.SO_REUSEADDR, 1)
    s.bind(addr)
    s.listen(2)
    print("Listening, connect your APP to http://%s:8080/" % ipAddress)

    counter = 0
    while True:
        res = s.accept()
        client_sock = res[0]
        client_addr = res[1]
        print("Client address:", client_addr)
        print("Client socket:", client_sock)

        req = client_sock.recv(1024)
        print("Payload: %s" % req.decode())
        client_sock.send(CONTENT % counter)
        client_sock.close()
        counter += 1
        print()

loop = asyncio.get_event_loop()
loop.create_task(toggle())
loop.create_task(webServer('192.168.4.1'))
loop.run_forever()
```

peterhinch, 2018-02-01:
Unfortunately `socket.getaddrinfo()` is a blocking method. A nonblocking DNS lookup is needed. This is a known issue but I don't think anyone is currently working on it. So we're stuck with that problem.

Note that by default sockets are blocking, so as written your `client_sock.recv` and `client_sock.send` will block, for periods dependent on the internet connection and the responsiveness of the site. It is possible to set them to be nonblocking. My asynchronous MQTT [library](https://github.com/peterhinch/micropython-mqtt.git) uses nonblocking sockets, so if you look at the code for **mqtt_as.py** you can pick up some pointers on the use of nonblocking sockets.

Sunrise17, 2018-02-01:
I have added `s.setblocking(False)` as seen in mqtt.py, under line of `s = socket.socket()`, unfortunately, it gives an error `Error in Wlan connect: [Exception] OSError: [Errno 11] EAGAIN`.

Sunrise17, 2018-02-01:
Dear @peterhinch , Is it possible to create two different loops? One for LED, another is for webServer?

Sunrise17, 2018-02-01:
With  asyncio.start_server method solved the issue, but one thing is i could not find a way overwrite the port which we did in socket method as `s.setsockopt(socket.SOL_SOCKET, socket.SO_REUSEADDR, 1)`
[https://stackoverflow.com/questions/48506460/python-simple-socket-client-server-using-asyncio/48507121#48507121](url)

Sunrise17, 2018-02-01:
With using awrite() method, finally i am not getting any error but client side has not succeeded to get the response from server.

Sunrise17, 2018-02-01:
'yield from' allow awrite method to send the response successfully.
`yield from writer.awrite("HTTP/1.0 200 OK\r\n\r\nMessage is received\r\n")`

I have only one remaining issue is, soft reboot does not allow board to start 'asyncio.start_server', always it is neccessary to reset() the board to start successful server. This problem is same like in socket but in socket method we can solve with this method `s.setsockopt(socket.SOL_SOCKET, socket.SO_REUSEADDR, 1)`, in asyncio how we can solve it?

peterhinch, 2018-02-02:
What platform are you running on? ESP32 still has some rough edges.

Sunrise17, 2018-02-02:
I am running on ESP8266 (Sonoff Basic device).

peterhinch, 2018-02-02:
You might be interested in [this forum post and thread](https://forum.micropython.org/viewtopic.php?f=16&t=3777&p=22204).

tl;dr I spent many weeks trying to get reliable operation from Sonoff units. And failed. This may or may not be relevant to your current issue, but it could save you a lot of wasted time.

Sunrise17, 2018-02-02:
Thanks for the feedback in the forum Mr. Hinch, nowadays I am using mqtt.py (without async), and that's fine at least for right now, but of course after i have made many tests, i will share the results. In my other units, Tasmota [(https://github.com/arendst/Sonoff-Tasmota)](url) is installed on them, and they are working also reliable for many days. Finally its different platform, i can not compare its architecture with micropython. Hope not only with Sonoff but also in other boards, to get a good results with Micropython (especially with SSL connections).

Sunrise17, 2018-02-07:
Dear @peterhinch , i have one question regarding to break the while True loop in async method. Here is my code, and it works successfully. But i want to add such property if its possible. For ex. 1st request is coming then colorpicker() method is activated (its not loop method), and then when i send 2nd request, wipeReturn() method is activated (this is loop method). Here is my issue, then when i send 1st request again, wipeReturn() should be stopped (give break), i want him to be stopped or paused. How can i apply that?

```
async def handle_client(reader, writer):
    req = await reader.read()
    if "a" in req.decode():
        yield from colorPick("128")
        yield from writer.awrite("HTTP/1.0 200 OK\r\n\r\n'Color: is applied'\r\n")
    elif "b" in req.decode():
        yield from wipeReturn(2)
        yield from writer.awrite("HTTP/1.0 200 OK\r\n\r\n'Color: is applied'\r\n")

async def colorPick(color):
    print('colorPick')
    np.fill(color)
    np.write()

async def wipeReturn(delay):
    while True:
        print('wipereturn')
        await asyncio.sleep(delay)

loop = asyncio.get_event_loop()
loop.create_task(asyncio.start_server(handle_client, ipAddress, 8080))
loop.run_forever()
```

closed 2018-02-07 by Sunrise17