peterhinch/micropython-async #15: How to get ISR from c to trigger asyncio event

readysteadygo2006, 2018-06-04:
Hi,

I'm developing an embedded system. I want to get C side ISRs to trigger asyncio 
activity. By doing so, I hope to avoid polling which drains batteries. 

I've understood how to get C side to call a python callback, but
can't get the python function to initiate an asyncio action.

My code is:
import uasyncio as asyncio
from uasyncio.queues import Queue

msgSentQueue = Queue()
sendMsgQueue = Queue()
msgRecvQueue = Queue()
cbQueue = Queue()
loop = asyncio.get_event_loop()

def callback(cbQueue):
    print('in Callback')
    cbQueue.put(1)
#    loop.call_soon(eventHandler)

async def eventHandler(cbQueue, msgSentQueue):
    #msg = ws.recvMsg()
    while True:
        result = await(cbQueue.get())  # may pause if a size limited queue fills
        msg = "test"
        print('in eventHandler')
        await msgSentQueue.put(msg)  # may pause if a size limited queue fills

async def cmdProc(msgRecvQueue):
    while True:
        result = await(msgRecvQueue.get())  # Will pause if q is empty
        print('Result was {}'.format(result))

async def bar(cbQueue):
    count = 0
    while True:
        count += 1
        print(count)
        callback(cbQueue)
        #await cbQueue.put(1)
        await asyncio.sleep(1)  # Pause 1s

I would like to call the function callback, from
c when an event occurs, but in my test I can't
get callback to initiate execution in eventHandler.

I've tried using both:
  cbQueue.put(1)
  loop.call_soon(eventHandler)
neither work.

However if I put 
        #await cbQueue.put(1)
in async function bar, all work
as expected!

Can you advice?!?!
BR
Paul

peterhinch, 2018-06-05:
Firstly I assume the Python callback is running when the interrupt occurs.

There are two ways to cause a coroutine to run. One is to **await my_coro()** (from a coroutine only, not a function). The calling coro pauses until the called one completes. The other, which may be issued from a coro or a function/method, is to issue **loop.add_task(my_coro())**. In a coro the choice between these depends on whether you want to pause until completion.

The methods **call_soon()** etc. should be used only to schedule functions/methods.

I think you need to use a synchronisation primitive to enable the event handler to know that an interrupt has occurred. If a queue is empty, doesn't **.get()** return immediately with nothing? An option is to have the ISR set an event and have the handler wait on it: by then it "knows" there should be something in the queue.

On the overall objective of power saving, this is not **uasyncio**'s forte, as the underling event loop runs continuously. You haven't mentioned what platform you're using but on the Pyboard the current draw is largely constant at ~70mA while **uasyncio** is running.

peterhinch, 2018-06-06:
I suggest you read the tutorial on **uasyncio** in [this repo](https://github.com/peterhinch/micropython-async.git) which has information on the MicroPython **asyncio** subset and also on how the event loop works. You can also inspect the code in [micropython-lib](https://github.com/micropython/micropython-lib.git).

closed 2018-11-30 by peterhinch