peterhinch/micropython-async #16: asyncio_priority

tvo321, 2018-06-06:
In trying to use a StreamReader (w/ uart) and a lower priority coro. It appears StreamReader's readline() never yields (the way it is implemented from examples I found).

Here's an example of what I'm referring to:
```
import asyncio_priority as asyncio
from pyb import UART
uart = UART(1, 115200)

async def reader():
    sreader = asyncio.StreamReader(uart)
    while True:
        res = await sreader.readline()
        print(res)

async def heartbeat(led):
    while True:
        led.toggle()
        await asyncio.after_ms(500)  # Will hang while a coro loops on a zero delay

loop = asyncio.get_event_loop()
loop.create_task(heartbeat(pyb.LED(4)))
loop.create_task(reader())
loop.run_forever()
```

I think a workaround would be to use the following (but is there a workaround to use StreamReader with your low priority coro library?):
```
async def reader():
    while True:
        if uart.any()> THRESHOLD: #check enough bytes has been received to ensure a '\n' 
            res = uart.readline() 
        await asyncio.sleep_ms(20)
```

peterhinch, 2018-06-07:
That's an interesting one, I'll investigate over the next few days.

There is a pending MicroPython PR #3836 which may interest you. I'm currently investigating a possible **uasyncio** issue where in a particular circumstance **StreamWriter** appears to stop **StreamReader** from running; this may or may not be related to your problem.

The PR removes some of the need for the high priority part of **uasyncio_priority**. A need for efficiently polling hardware was the original reason behind **uasyncio_priority** and the PR is causing me to wonder if it is worth continued support. When **uasyncio** changes, **uasyncio_priority** needs fairly radical and time-consuming modification. I'd be interested to hear the views of yourself (and any other users).

peterhinch, 2018-06-14:
OK, I've now had a chance to test this and can't replicate your problem. Perhaps there is a version issue?

Please ensure that you have the latest version of **uasyncio** and **uasyncio_priority**. Early this year uasyncio had a major change to version 2.0 and I had to make extensive changes t **uasyncio_priority** to match.

peterhinch, 2018-06-15:
I should add that, as stated in my README, **uasyncio** 2.0 requires firmware dated 22nd Feb 2018
or later.

Asynchronous programming can greatly simplify application design. Most firmware I have written over the years has used it, going back to the days of Intel 8080 Assembler ;-)

tvo321, 2018-06-15:
So I am running on a pyboard v1.1, and my firmware build is: 
MicroPython v1.9.4-130-ga8b9e71a on 2018-06-06, So I should be up to date there. 

I got uasyncio last week via a unix build and I used upip to get the library then copied the required files to the pyboard.

As for the application, the uart device I'm interfacing to sends 94 bytes of data every 50ms at 115200 baud. As soon as I physically disconnect the uart device, the heartbeat led blinks correctly. However, as soon as I plug the device back in, the heartbeat led "latches" at its last setting. Is this timing too demanding perhaps?

peterhinch, 2018-06-16:
It is possible. Assuming you're using the **StreamReader** and **StreamWriter** classes to access the UART there are ongoing discussions regarding the underlying design of **uasyncio** (which was not developed by myself). There are situations in which UART I/O can fail [see this discussion](https://github.com/micropython/micropython/pull/3836#issuecomment-397317408). I have a proposal for fixing this under consideration. I also have a proposal for raising I/O priority [discussed here](https://github.com/micropython/micropython/issues/2664). If this is accepted it will make my "priority" version effectively obsolete by providing a superior, built-in high priority I/O subsystem.

tl;dr uasyncio UART stream I/O at high baudrates is under considerable discussion. Full-duplex protocols have particular problems.

A suggestion. You might consider trying your code without my priority version and using uasyncio.sleep() for the LED: this might make it better able to compete with the I/O. If you experience character loss you might try increasing the UART buffer size. Aside from that it's possible that you are hitting the current limits of **uasyncio**. I would be very interested in hearing whether your application works with standard **uasyncio**. If it exhibits problems it might help bolster my case for raising I/O priority.

It's also possible that you've actually completely monopolised the CPU, depending on what you're doing with the data. 50ms isn't long and **uasyncio** isn't a magic bullet: if the CPU is running flat-out something will fail.

tvo321, 2018-06-26:
Ok, I've been able to run a few tests and just wanted to report my results. 

I developed more of my application. The application is communicating with several devices over UART (using 3 uarts on the pyboard). Data is received asynchronously from 2 serial devices, and the data collected is transmitted on a 3rd uart. I used 2 StreamReader objects to receive the incoming data, and a Streamwriter object to transmit the data back out. 

Using 4 co-routines: 

1) Streamreader object, uart 1 for device 1. 92 bytes of data at 115200 baud every 50ms (~8ms to transmit). The data is stored in a globally accessible variable as it is received (with a 'time-stamp' via utime.ticks_ms). 
2) Streamreader object, uart 3 for device 2. 40 bytes of data at 57600 baud every 25ms (~7ms to transmit). This data is also stored in a globally accessible variable as it is received (with a 'time-stamp' via utime.ticks_ms). 
3) Streamwriter object, uart 6 for device. Transmitting 132 bytes at 230400 baud (~5.7 ms to transmit). The globally stored data is read and printed out every 50ms via Streamwriter (essentially grabbing the latest globally accessible data from coros 1&2)
4) Heartbeat with an asyncio.sleep(5)

In this configuration, the data in coro 1 seems to come in consistently every 50ms. Where as the data in coro 2, seems to be inconsistent ranging anywhere from 50ms to 2000ms. 

However, without using the StreamReader object and "polling" the uart.any() with an asyncio.sleep_ms(10). Data is read from the uart, when the uart.any() > 92 (for device 1) and when uart.any() > 40 (for device 2) is met. 

Device 1 receives data every 50ms.
Device 2 receives data every 38ms. 

I also attempted using the _thread module, also doing a similar polling style with uart.any() in each thread, with consistent results as well.

peterhinch, 2018-06-27:
I have [this PR](https://github.com/micropython/micropython-lib/pull/287) which provides a way of increasing the priority of I/O relative to tasks which yield with a zero delay. However from your description none of your tasks do this so I don't think it will help, but feel free to try it. The I/O performance of **uasyncio** is currently poor when it competes with zero delay tasks. But if all tasks wait on a finite delay, it is good.

Aside from that I'm not sure what I can offer: you may simply be hitting the limits of practical performance, whether of **uasyncio** or the processor.

In general, as you probably know, **_thread** is more resource-intensive than **uasyncio**.

peterhinch, 2018-11-30:
I assume this is now fixed. Please open a new issue if not.

closed 2018-11-30 by peterhinch