peterhinch/micropython-async #17: running a aremote.py on esp32

fruch, 2018-06-12:
I was running aremote.py few months back on esp8266, I have the new esp32 and tried using it.
got it to run, but it wasn't working, i.e. not of my of my remote click weren't cought and didn't recived on the _pin_cb function.

I tird wiring in every single combo, but it wasn't working. is there some limitation with esp32 input pin ? or something else i'm missing here ?

peterhinch, 2018-06-13:
I haven't done very much on ESP32. I'm confident that **uasyncio** works on the platform so I think the most likely cause is the fact that **aremote** relies on a pin interrupt. I suggest you write a simple test script that does something like toggling an LED on a pin change to verify that pin IRQ's work. The ESP32 port is much less mature than the ESP8266 so it's possible that there are issues or limitations on pin IRQ's. If your script doesn't work a general query about pin IRQ's on [the forum](https://forum.micropython.org/) should produce a knowledgeable response.

Assuming I'm right about the IRQ the only line in the code which may require change is [this one](https://github.com/peterhinch/micropython-async/blob/48d1d52b922680b28644acdb65d4465b4c617c16/nec_ir/aremote.py#L52) which initialises the pin ISR.

I haven't time to investigate this myself at the moment but I'd appreciate it if you could report the outcome to help me fix the code.

WimRoose, 2018-07-30:
@peterhinch @fruch I can confirm it works on ESP32 (LOLIN D32 v1) - like @peterhinch mentioned, except for the platform checks you only have to change the line that initializes the pin to

`pin.irq(handler = self._cb_pin, trigger = (Pin.IRQ_FALLING | Pin.IRQ_RISING))`

(so remove hard= true), and change pin number of course

peterhinch, 2018-07-30:
Driver and demo scripts now updated. Tested under official firmware but they also check for the Loboris port so should be compatible.

closed 2018-07-30 by peterhinch

GHPS, 2020-01-17:
> Driver and demo scripts now updated. 

The driver now runs great on the ESP32 platform -
thanks for your work!

The respective README file, however, explicitly mentions only
the ESP8266 which in most projects means that the code
is limited to the later.

It would be helpful if you would update the README 
to reflect the recent changes.

peterhinch, 2020-01-18:
Thanks for pointing this out. I've updated my local copy. This entire repo is in the course of a major update to reflect the forthcoming rewrite of official **uasyncio**. It will be a while before the changes appear as there will be associated code changes.

GHPS, 2020-01-19:
> This entire repo is in the course of a major update to reflect the forthcoming 
> rewrite of official uasyncio.

I'm looking forward to seeing these updates - both the rewrite of uasyncio and
your tutorials.

Pythons asyncio on microcontrollers is such a perfect match that after a
short while one wonders how the development of more complex programs
was even possible without it. For example: Debouncing a switch => read
the state and await the debounce time. Elegant, productive and even easy 
to read...

And your introductions are really helpful getting started with uasyncio.
For beginners (and who has a profound knowledge with asyncio and
microPython?) they are even essential!