peterhinch/micropython-async #18: Cancellable for Function with no argument .

curlyz, 2018-07-25:
Hi , 
def cancellable(f):
    def new_gen(*args, **kwargs):
        if isinstance(args[0], TaskId):  # Not a bound method
            task_id = args[0]
            _g = f(*args[1:], **kwargs)_
        else:  # Task ID is args[1] if a bound method
            task_id = args[1]
            args = (args[0],) + args[2:]
            g = f(*args, **kwargs)
        try:
            res = await g
            return res
        finally:
            NamedTask._stopped(task_id)
    return new_gen

Apparently this will raise TypeError when the function doesnt require any arg .  
Example .
`
num = 1
@asyn.cancellable
async def print_nums():
    while True:
        global num;print(num)
        led.value(not led.value())
        num += 1
        await asyn.sleep(0.1)
`

Sorry for bad copy , no idea how this work , the problem is on line 327 of asyn.py .

peterhinch, 2018-07-25:
I can't replicate this. I created a modified version of cantest.py with test4 changed so that cant40 takes no args. It runs.
```python
@asyn.cancellable
async def cant40():  # Now takes no args
    num = 1
    while True:
        try:
            await asyn.sleep(1)
            print('Task cant40 no. {} running.'.format(num))
        except asyn.StopTask:
            print('Task cant40 no. {} was cancelled'.format(num))
            return

@asyn.cancellable
async def cant41(num, arg=0):
    try:
        await asyn.sleep(1)
        print('Task cant41 no. {} running, arg {}.'.format(num, arg))
    except asyn.StopTask:
        print('Task cant41 no. {} was cancelled.'.format(num))
        return
    else:
        print('Task cant41 no. {} ended.'.format(num))

async def cant42(num):
    while True:
        print('Task cant42 no. {} running'.format(num))
        await asyn.sleep(1.2)

# Test await syntax and throwing exception to subtask
@asyn.cancellable
async def chained(num, x, y, *, red, blue):
    print('Args:', x, y, red, blue)  # Test args and kwargs
    try:
        await cant42(num)
    except asyn.StopTask:
        print('Task chained no. {} was cancelled'.format(num))

async def run_cancel_test4():
    await asyn.Cancellable(cant41, 0, 5)
    loop = asyncio.get_event_loop()
    loop.create_task(asyn.Cancellable(cant40)())  # 3 instances in default group 0
    loop.create_task(asyn.Cancellable(cant40)())  # No args passed
    loop.create_task(asyn.Cancellable(cant40)())
    loop.create_task(asyn.Cancellable(chained, 4, 1, 2, red=3, blue=4, group=1)())
    loop.create_task(asyn.Cancellable(cant41, 5)())  # Runs to completion
    print('Running tasks')
    await asyncio.sleep(3)
    print('About to cancel group 0 tasks')
    await asyn.Cancellable.cancel_all()  # All in default group 0
    print('Group 0 tasks were cancelled')
    await asyncio.sleep(1)  # Demo chained still running
    print('About to cancel group 1 tasks')
    await asyn.Cancellable.cancel_all(1)  # Group 1
    print('Group 1 tasks were cancelled')
    await asyncio.sleep(1)
```

curlyz, 2018-07-28:
Thanks Peter , it turn out that I misunderstand async operation ..
I am wonder if it is necessary that you must both use decorator for function 
`
@asyn.cancellable
`
and also 
`loop.create_task(asyn.Cancellable(cant40)())`

peterhinch, 2018-07-29:
I suggest you study `cantest.py` and the code sample above. You'll also find a simple example in `asyntest.py`. The decorator is used only on a task definition where that task is to be cancellable.

closed 2018-08-01 by curlyz