peterhinch/micropython-async #19: Make the cancellable condition recursive through await functions

goatchurchprime, 2018-08-15:
My understanding is that you have one subset of coros in the pool which have been assigned the cancellable property, and these all get stopped when we issue the cancel_all().  

But I'd really like an emulation of the behavior of threads where, once execution has entered via the entry point, functions call other functions as they normally do, but the entire stack can be killed back to this entry point via a thread.stop.  

Imagine if we had a single Cancellable declaration on a function that was made using a create_task() function (so we can think of the create_task() as a bit like a thread.start() at that function), and any call from within that function using the await keyword inherited the cancellable condition and belonged to the same sub-pool).  Then I could think of this whole set as its own thread, which I could control and kill off all at once at my convenience.  

My use-case is here:
    https://github.com/goatchurchprime/jupyter_micropython_developer_notebooks/blob/master/scriptlet_technology/MQTT_scriptlets.ipynb

I am asyncronously waiting for an MQTT message (broken into pieces in this case and stitched together into a file called scriptlet.py) containing a piece of python code, which I execute asyncronously, and then continue waiting on the MQTT channel for more sets of python code while it continues to execute.  This is where I had reason to want to kill off all the previously running functions received earlier.  

I'd like not to have to insert @cancellable onto every single function in the scriptlet; it's enough to need to put in async and await everywhere (and maybe I could do that with some preprocessing, so I could pretend the same code can run natively as within this async environment).

peterhinch, 2018-08-16:
Sending scripts via MQTT is a nice idea.
> and any call from within that function using the await keyword inherited the cancellable condition and belonged to the same sub-pool

This is already the case: the `CancelledError` propagates back to the calling coro as can be demonstrated by pasting this script:
```python
import uasyncio as asyncio
import asyn

async def forever(n):
    print('Started forever() instance', n)
    while True:
        await asyncio.sleep(1)
        print('Running instance', n)

@asyn.cancellable
async def rats(n):
    await forever(n)

async def run_cancel_test2():
    loop = asyncio.get_event_loop()
    loop.create_task(asyn.Cancellable(rats, 1)())
    await asyncio.sleep(5)
    print('About to cancel tasks')
    await asyn.Cancellable.cancel_all()
    print('Tasks were cancelled. Pause 3 secs to prove it.')
    await asyncio.sleep(3)

loop = asyncio.get_event_loop()
loop.run_until_complete(run_cancel_test2())
```
Perhaps what you intend is the case where a cancellable task launches a coro rather than awaiting it:
```python
async def foo():  # No decorator
    await something_or_other()

@asyn.cancellable
async def bar(n):
    loop = asyncio.get_event_loop()
    loop.create_subtask(foo())  # foo becomes part of bar's cancellable group
```
I can see that this is convenient for your use case but I'm unconvinced it adds functionality of general value. I've not yet thought through the implementation; if I spot a low cost solution I'll consider it.

peterhinch, 2018-11-30:
On reflection I think it would be a great feature but implementation would require a significant redesign of **uasyncio**, and would incur a performance penalty. For every task **uasyncio** would need (recursively) to store a reference to each subtask. So when a task receives **CancelledError** it recursively propagates it to all subtasks.

I would handle this at the application level using groups of cancellable tasks. It's not as elegant but it preserves the performance of **uasyncio**.

closed 2018-11-30 by peterhinch