peterhinch/micropython-async #21: Possible reliability problem with aswitch debounce mechanism

kevinkk525, 2018-08-17:
When using your aswitch module as underlying basecode I realized, that you are not using the debounce in the normal way. Maybe your intention was to improve response time, I don't know.

Expected behaviour: 
If the pin state changes, wait the debounce time, then check the state again. If it is still the same, trigger the callback.

Actual behaviour:
If the pin state changes, immediately trigger the callback, then wait the debounce time, which in this case actually is just a delay until the state is checked again.

Expected problems:
Mainly a reliability problem. The pin state change could be a very short burst caused by fluctuations on the power line.
Example: The bell system in my house is very old. If someone rings the bell of any of my neighbours, I get a short burst on my bell line too. Not enough to trigger the real bell of course but with my esp8266 without a long enough debounce time I can see that burst and it would trigger the callbacks.
Therefore I'd have a reliability problem using your debounce implementation with my bell system.

Suggestion:
Check pin state after the debounce time again to ensure that it was not only a fluctuation/very short spike.

peterhinch, 2018-08-18:
The intention is indeed to minimise response time, for example in games. The code claims only to handle contact bounce with switches connected to the host via short wires. It isn't a low pass digital filter.

For switches connected over long lines where interference is a potential issue it is normal to have analog filtering prior to the digital input, firstly to remove high frequency interference and also to protect the input from transient overvoltage. In systems such as intruder alarms it's not unknown to use optical isolation to provide protection. In short, you're entering the realms of electronic engineering, where there is a range of established techniques for constructing line receiver circuits.

I have no plans to alter this module which (in my view) serves its design purpose. You are, of course, free to implement code as you describe to provide a degree of digital filtering at the cost of response time. However I would regard such filtering as only one component of a lie receiver solution.

kevinkk525, 2018-08-18:
Thank you for your answer.
This makes sense to me, thanks for the explanation.
I understand now that the code is perfectly fine for that purpose.

closed 2018-08-18 by kevinkk525