peterhinch/micropython-async #22: aswitch with both short and long press

pacmac, 2018-11-28:
Just started using your asyncio libs a few hours ago and are trying to implement the Pushbutton press and long functions.

```
    pb = Pushbutton(gpio.but)
    pb.press_func(self.butshort,())
    pb.long_func(self.butlong,())
```

Problem is, when it is a long press, first the press fires and then the long fires.

How can I configure this so that only long fires  during a long press ?

peterhinch, 2018-11-28:
This is by design. The reason is that fixing it would introduce latency in the short press case (it necessarily takes time to determine that the press is not long). Such latency would be problematic in many applications (e.g. games).

One option is to use the button release which occurs after the long/short status has been established. The other is to adapt the class to suit your needs.

peterhinch, 2018-11-28:
This is now done (and documented) in branch longpress.

I'd appreciate your tests/comments before I merge it.

diginfo, 2018-11-29:
Hi Peter;

Thanks, yes, tested and working.

Actually I prefer for my long-press to tell the user when the long-press time has exceeded (normally with a beep) as otherwise there's uncertainty as to how long they have to press for short and long, it makes a huge improvement in usability as fed back from users.

The user holds down the button and a double-beep tells them to let go, if they release early (short press) then I use a single beep.

I have tried to convert existing firmware that uses interrupts and a main loop by using your fast_io async and other libs, but being new to this it's not behaving, but that's not doubt due to my lack of experience with asyncio.

peterhinch, 2018-11-29:
Please note I'm still working on this and the API may change. I'm trying to generalise this to cope with double clicks as well as long presses so that (optionally) we have an event which occurs only if neither a double click nor a long press has occurred. I'm thinking that the conditional event should be the release function rather than the press function.

@diginfo Asynchronous programming takes some time and effort to learn, but it's well worthwhile.

pacmac, 2018-11-29:
I have this function that is self explanatory:

```
  async def beep(self,rep=1,dur=10,pause=50):
    for i in range(rep):
      gpio.bpin.value(1)
      await asyncio.sleep_ms(dur)
      gpio.bpin.value(0)
      await asyncio.sleep_ms(pause)
```

But how would I call this as calling it normally just fails silently as it says it would in your tutorials ?

peterhinch, 2018-11-29:
I've now pushed an update which has passed my tests - hopefully the API will now be stable.

The button press function always occurs immediately. The behaviour of the button release function now depends on a constructor arg, now renamed `suppress`. If this is set, the button release function is only launched if neither a long or doubleclick event occurs. This is described in more detail in the README.

Shifting it to the release function made sense as the soonest possible detection of the absence of a long press is on button release. I also have a suspicion that an unconditional low latency button press event may have application even if another function is suppressed.

Comments welcome!

peterhinch, 2018-11-30:
Now merged.

closed 2018-11-30 by peterhinch