peterhinch/micropython-async #26: Handing loop object to function/class

TrevisGordan, 2019-09-03:
Not a real issue:
Is there a difference when handing the loop object to the function or class, similar to Test 5&6? 
I always thought there is only one loop instance. But now I am unsure. And since `asyncio.Task.all_tasks()` is not available I can't find it out.

https://github.com/peterhinch/micropython-async/blob/8239a3d8b9f19430a13606c1492d5bf9a18300eb/cantest.py#L422

peterhinch, 2019-09-04:
There is only one instance.

The way `uasyncio` works is that the event loop is instantiated on the first execution of
```python
loop = asyncio.get_event_loop()
```
This is Paul's way of implementing singleton objects. It is efficient, but it has a drawback. You can, at any time access an already instantiated event loop quickly and efficiently by issuing
```python
loop = asyncio.get_event_loop()
```
On the face of it this avoids the need to pass the `loop` instance around your program. But there's a "gotcha".

The drawback occurs if, at the start of your main module, you import a module which does this. Your code might attempt to instantiate the event loop with non-default args, but by the time you do this the module has already instantiated it using the defaults. Unless you know how **uasyncio** works this is rather a subtle bug.

There are two workrounds. One is to issue `get_event_loop(args)` after importing **uasyncio** but before importing other modules. The other is to design classes, methods and functions to require `loop` as an arg. I prefer the second because, however the user code is written, the error cannot occur; but if the module you're importing doesn't use this convention you need to fall back on the first.

Alas this hadn't occurred to me when I first released the `asyn` and `aswitch` modules. Rather than break people's code I haven't altered them to conform to this.

My `fast_io` fork implements `get_running_loop()` from Python 3.7 to enable modules to throw a `RuntimeError` if the event loop is not instantiated.

TrevisGordan, 2019-09-07:
Thanks a lot! This made me think a lot for a while now. In which form does this error occur? Cause it will not throw a right?

peterhinch, 2019-09-07:
Calling `get_event_loop` with args will not throw an exception. The args will only be used on the first call after a reboot; subsequently they will silently be ignored. The solutions/workrounds are as I described above.

I considered modifying my uasyncio fork to addresss this, but decided against. My overriding aim was to avoid breaking working code (even if poorly configured).

closed 2019-09-09 by TrevisGordan