peterhinch/micropython-async #27: IndexError While using the "Chain" Method on a function with @asyn.cancellable

TrevisGordan, 2019-09-03:
Now I have a real issue:
If you use the normal await style from the chain example e.g:
`result = await compute(x, y)` 
on a function with an `@asyn.cancellable` decorator, it will throw an: 
`IndexError: tuple index out of range` Error.
(File "asyn.py", line 327, in new_gen)

https://github.com/peterhinch/micropython-async/blob/8239a3d8b9f19430a13606c1492d5bf9a18300eb/chain.py#L15

https://github.com/peterhinch/micropython-async/blob/8239a3d8b9f19430a13606c1492d5bf9a18300eb/asyn.py#L327

peterhinch, 2019-09-04:
I think you'll have to show me your code. The following works as expected (tested on a Pyboard D SF2W):
```python
try:
    import uasyncio as asyncio
except ImportError:
    import asyncio
import asyn

@asyn.cancellable
async def compute(x, y):
    print("Compute %s + %s ..." % (x, y))
    await asyncio.sleep(1.0)
    return x + y

async def print_sum(x, y):
    result = await compute(x, y)
    print("%s + %s = %s" % (x, y, result))

loop = asyncio.get_event_loop()
loop.run_until_complete(print_sum(1, 2))
loop.close()
```

TrevisGordan, 2019-09-07:
Thanks for the quick response, I somehow missed it.
I Sure will show you the codes.

While trying to reproduce the error again I found another. 
Note: The following test were made on ESP32 and also ESP8266

1. If I run you above code as it is, I receive an: `TypeError: function takes 2 positional arguments but 1 were given`

see output:
```
MicroPython v1.11-273-g8fc00928e on 2019-09-04; ESP32 module with ESP32
Type "help()" for more information.
>>> import hinch
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "hinch.py", line 18, in <module>
  File "uasyncio/core.py", line 181, in run_until_complete
  File "uasyncio/core.py", line 110, in run_forever
  File "uasyncio/core.py", line 178, in _run_and_stop
  File "hinch.py", line 14, in print_sum
  File "/Users/xx/xx/xx/asyn.py", line 333, in new_gen
TypeError: function takes 2 positional arguments but 1 were given
```

2. The mentioned `IndexError: tuple index out of range` Error, only is thrown when the awaited function has no arguments.

If I ran this code ill get the mentioned IndexError.

```python

# From:
# chain.py Demo of chained coros under MicroPython uasyncio
# Author: Peter Hinch
# Copyright Peter Hinch 2017 Released under the MIT license

# peterhinch Github Issue: #27
# Test to reproduce: IndexError: tuple index out of range

# Note: Error only apears when NO Arguments are past to the awaited function.

try:
    import uasyncio as asyncio
except ImportError:
    import asyncio
import asyn

@asyn.cancellable
async def compute():
    # No args are given to the compute function.
    print("compute function")
    await asyncio.sleep(1.0)
    return 1

async def print_sum(x, y):
    print("compute function")
    result = await compute()  # no args are passed
    print(result)

loop = asyncio.get_event_loop()
loop.run_until_complete(print_sum(1, 2))
loop.close()
```
With folling output:

```
MicroPython v1.11-8-g48dcbbe60 on 2019-05-29; ESP module with ESP8266
Type "help()" for more information.
>>> import issue27.py
print_sum
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "issue27.py", line 32, in <module>
  File "uasyncio/core.py", line 180, in run_until_complete
  File "uasyncio/core.py", line 154, in run_forever
  File "uasyncio/core.py", line 109, in run_forever
  File "uasyncio/core.py", line 177, in _run_and_stop
  File "issue27.py", line 28, in print_sum
  File "/Users/xx/xx/xx/asyn.py", line 327, in new_gen
IndexError: tuple index out of range
```

I hope this helps.

peterhinch, 2019-09-07:
You are using the module incorrectly. There is a required syntax for calling an awaitable function. In the no-args case it is
```python
try:
    import uasyncio as asyncio
except ImportError:
    import asyncio
import asyn

@asyn.cancellable
async def compute():
    # No args are given to the compute function.
    print("compute function")
    await asyncio.sleep(1.0)
    return 1

async def print_sum(x, y):
    print("compute function")
    result = await asyn.Cancellable(compute)()
    print(result)

loop = asyncio.get_event_loop()
loop.run_until_complete(print_sum(1, 2))
loop.close()
```
Or when passing args it is:
```python
try:
    import uasyncio as asyncio
except ImportError:
    import asyncio
import asyn

@asyn.cancellable
async def compute(x, y):
    # No args are given to the compute function.
    print("compute function")
    await asyncio.sleep(1.0)
    return x + y

async def print_sum(x, y):
    print("compute function")
    result = await asyn.Cancellable(compute, x, y)()
    print(result)

loop = asyncio.get_event_loop()
loop.run_until_complete(print_sum(1, 2))
loop.close()
```
This isn't the first time this has caused confusion - I'll try to find time to revisit my docs.

Incidentally my **fast_io** fork fixes this bug in **uasyncio V**2. Tasks may be cancelled in the normal way without the need for special syntax. Paul Sokolovsky's version fixes this bug in a more efficient way but it requires his **pycopy** firmware; mine uses official firmware.

TrevisGordan, 2019-09-09:
Yes, I perfect I think the warning should suffice. Thank you! 
I found out about pycopy a few days ago and so far not sure, about using it, wanted to get my head around the "Vanilla" version first. 
Thank you again for your incredible work on those topics!

closed 2019-09-09 by TrevisGordan

peterhinch, 2019-09-09:
> I found out about pycopy

This is your choice. I only support official MicroPython firmware for a few reasons, not least because binaries are available to those who are reluctant or unable to build from source.