peterhinch/micropython-async #30: Calling `aclose()` on StreamReader ?

DrTom, 2019-11-14:
When closing a StreamReader e.g. as in: 
```
 sr, sw = await asyncio.open_connection(host, port, ssl=use_ssl)
 # ....
 await sr.aclose()
```
the program will crash. My code runs also on `cpython` and interestingly the corresponding StreamReader on that platform doesn't even have a `.close()` method. 

So my the questions is: am I even supposed to call `sr.aclose()` (and  if not why does it exist)?

peterhinch, 2019-11-15:
I think the method should not exist.

Are you running official uasyncio V2 or my **fast_io** fork? If you are running **fast_io** I suggest you re-test with the official version. If the official version fails, please could you raise an issue against MicroPython as I am not a maintainer for the official version.

If only **fast_io** fails, please report back and I will remove the method.

DrTom, 2019-11-15:
I installed asyncio via `upip` as you suggested in your tutorial, which is very helpful by the way. So it is not your "fast" version.  I am not sure if it is the "official" version because https://pypi.org/ points to https://github.com/pfalcon/pycopy-lib via its homepage link. I just found that out and so I am a bit confused because I assumed that `upip` would pull the official version. Those micropython forks are a bit disorienting for people new to the eco system (like me). 

Anyways, I write my code portable between cpython and micropython and the mentioned new uasyicio seems to make this easier. I will give it a try once released.

peterhinch, 2019-11-16:
`upip` does pull in the official version, from a repository other than PyPi. You may want to raise an issue against MicroPython - that is up to you. I gather the existing version will be retained under a new name.

In my opinion, having been involved in testing, the new version is excellent. It includes features from CPython 3.8 and a primary design aim is improving CPython compatibility.

In due course I plan to change this repo to support the new version. The tutorial should only need minor changes.

DrTom, 2019-11-21:
I am fairly sure now, that used library originaly stems from https://github.com/pfalcon/pycopy-lib and I as far as I can reconstruct it was pulled via `upip`.  I will keep it for now. I  disabled `await` on the reader https://github.com/DrTom/py-u-async-http-client/blob/ee7a25a509c13a9db35acc6375c2ce12b4816a62/http_client/core.py#L40 . I plan to migrate to the new asyncio version as suggested. At any rate has the problem nothing to do with this project and I am closing this issue.

closed 2019-11-21 by DrTom