peterhinch/micropython-async #31: OPENMV 4   AttributeError: 'module' object has no attribute 'deque'

skylin008, 2020-02-23:
Hi, had coording to the guide doc copy the related file to asyncio folder . When I run the auart.py on OPENMV4,but any error as show: 'module' object has no attribute 'deque',how to solve this issue.The OPENMV4 firmware is :  MicroPython v1.11-omv OpenMV v3.5.1 2019-12-06; OPENMV4-STM32H743
Thanks!

peterhinch, 2020-02-23:
This is an issue with the openmv firmware build, and implies it is unable to run official uasyncio. This could be taken up with openmv. 

The file **core.py** in uasyncio uses **ucollections.dequeue**, and this is provided by the file **py/modcollections.c** in the source tree. If you have access to the openmv source tree (I don't know if this is available) you might want to investigate.

skylin008, 2020-02-23:
Thanks for your kindly advice,I will be try!

closed 2022-11-25 by peterhinch