peterhinch/micropython-async #32: AttributeError: cancel (ESP32)

gilphilbert, 2020-03-27:
I'm trying to cancel a task created using a colo. The task runs and the start_progress() function runs, but the cancel fails as the board reports that the cancel attribute doesn't exist.

I'm running the latest 3.x firmware (`esp32-idf3-20200327-v1.12-308-gdbba6b05d.bin`) on an ESP32 (Adafruit ESP32 Feather).

Any hints? I can't figure out what I'm doing wrong.

This is essentially my code, with the irrelevant parts removed.
```
import uasyncio as asyncio

async def start_progress():
    print('test')
    await asyncio.sleep(1)

async def main(loop):
    #get data from websocket...
    if data.get('state') == 'play':
        progress = start_progress()
        loop.create_task(progress)
    else:
        asyncio.cancel(progress)

loop = asyncio.get_event_loop()
loop.create_task(main(loop))
loop.run_forever()
```
If you want the full code, let me know and I'll point you to it.

Thanks!

peterhinch, 2020-03-27:
I think this is because you are using the new version of **uasyncio**. The **cancel** method is now a method of the **Task** instance so your code should read
```python
progress.cancel()
```
I suggest you look at [the new docs](http://docs.micropython.org/en/latest/library/uasyncio.html).

I have a substantial task updating my repo to match the new **uasyncio** so there will be a few discrepancies for a while. Sorry.

gilphilbert, 2020-03-27:
Ah, you know I read that in the docs but made that deadliest of flaws... I _assumed_ that I was on the the older version!

I'll try later and report back.

closed 2022-11-25 by peterhinch