peterhinch/micropython-async #39: Question: async write file to SD card

dukeduck1984, 2020-07-08:
Hi Peter,

Thanks for the tutorial.

In my case, I have a displacement sensor which is an encoder triggered by pulling the cable.  I need to write the data to SD card when certain conditions are met (e.g. some movement or displacement which are in a random matter).  Since writing to SD card can be slow, I'm afraid the writing process will block the sensor from getting new data.  How can I make it non-blocking with uasyncio V3?  Would you pls give me an example?  Thanks.

Kaiyuan

dukeduck1984, 2020-07-09:
This is my try, is it the right way to do it for uasyncio V3?

```
@asyn.cancellable
async def _save_it(self):
    with open(self.file,'w') as f:
        ujson.dump(self.config, f)
        await asyncio.sleep_ms(100)

async def save_new_config(self, new_value):
    self.config['the_key'] = new_value
    await self._save_it()
```

peterhinch, 2020-07-09:
Hi Kaiyuan,
unfortunately this isn't possible with the current filesystem drivers because these are based on synchronous code. Asynchronous I/O requires an asynchronous device driver. Short of writing asynchronous drivers, your only options are to cache the data for writing out later or to send the data elsewhere for storage.

Data could be cached in RAM or possibly in some fast device such as a [FRAM](https://github.com/peterhinch/micropython_eeprom/blob/master/fram/FRAM.md). This is a synchronous driver, but it has an option to access data as bytes: this will be faster than filesystem access. And the hardware doesn't suffer from the slow erase time of flash devices.

Having said that, SD cards have their own microcontroller. Perfromance depends on the speed grade of the SD card. So start out by measuring the blocking time of filesystem access to your fastest SD card to see if there's a practical problem.

Cheers,
Pete

dukeduck1984, 2020-07-09:
Thanks for the explaination Peter.  So for the current filesystem driver, once the writing starts, it has to wait until it's finished and the file's closed, and my code above won't work as how I intended to, is that correct?

dukeduck1984, 2020-07-09:
Maybe I can try using `_thread` -  having a while True loop keep watching a variable, whenever some data are passed to that variable, it will perform writing, and then set that variable back to None and keep watching for the next incoming data....

peterhinch, 2020-07-09:
@dukeduck1984 uasyncio V3 is available now. It is incorporated in daily builds and I recommend that you use it. You'll find resources and a new tutorial in the v3 directory. V3 removes the need for the **@asyn.cancellable** decorator: this was a workround for a major bug in uasyncio V2. Task cancellation works properly and in a CPython compatible manner in V3.

Re file I/O the lines
```python
    with open(self.file,'w') as f:
        ujson.dump(self.config, f)
```
will block for a period which will depend on the quality of the "disk" media. While your code wouldn't break, there is no benefit in wrapping it in uasyncio. The following synchronous function will be just as good (or bad, depending on your POV).
```python
def _save_it(self):
    with open(self.file,'w') as f:
        ujson.dump(self.config, f)
```
If the blocking period is excessive then threading may help. I have no experience of threading under MicroPython. Clearly the threading subsystem will grab timeslices from your main application. The question is whether those timeslices are longer than the blocking time of the simple synchronous _save_it function. I don't know the answer to that. You could experiment, or ask in the forum.

dukeduck1984, 2020-07-09:
Thanks so much for the help Peter.

I'm closing the issue.

closed 2020-07-09 by dukeduck1984