peterhinch/micropython-async #45: asyncio.get_event_loop() is not attribute time()

danielcruzmx, 2020-08-28:
File "aremote.py", line 80, in _cb_pin
AttributeError: type object 'Loop' has no attribute 'time'

79                loop = asyncio.get_event_loop()
80                 self._ev_start.set(loop.time())  # asyncio latency compensation

>>> import uasyncio as asyncio
>>> obj = asyncio.get_event_loop()
>>> dir(obj)
['__class__', '__module__', '__name__', '__qualname__', 'close', 'stop', '__bases__', '__dict__', 'create_task', 'run_until_complete', 'call_exception_handler', '_exc_handler', 'run_forever', 'set_exception_handler', 'get_exception_handler', 'default_exception_handler']

no time

peterhinch, 2020-08-28:
You are running the V2 driver with V3 **uasyncio**. Please run [this version](https://github.com/peterhinch/micropython-async/blob/master/v3/as_drivers/nec_ir/aremote.py). Note that it is now a Python package: see [the docs](https://github.com/peterhinch/micropython-async/blob/master/v3/docs/NEC_IR.md).

danielcruzmx, 2020-09-01:
Thanks, Additionally I added the primitives library and the art.py test worked.

closed 2020-09-01 by danielcruzmx