peterhinch/micropython-async #46: check_async_code.py

IhorNehrutsa, 2020-09-08:
https://github.com/peterhinch/micropython-async/blob/master/check_async_code.py - page not found.

peterhinch, 2020-09-08:
[Now in V2 directory](https://github.com/peterhinch/micropython-async/blob/master/v2/check_async_code.py) I haven't yet checked that utility against usyncio V3 code.

Where did you find the link, so I can fix it?

IhorNehrutsa, 2020-09-08:
Thanks. 
But you can't fix it. That is a translation to Russian your tutorial. There is a link to the author under the title. 
https://habr.com/ru/post/484472/
Maybe you might restore an original V2 location?

peterhinch, 2020-09-09:
Version 2 of uasyncio is essentially obsolete, and I plan to delete the V2 directory and its contents when uasyncio I/O performance is improved. This is promised for the next firmware release 1.14.

Before then I will adapt check_async_code.py for V3 and put it in the V3 directory.

More importantly, the Russian translation appears to be of my V2 tutorial. There are significant differences in uasyncio V3 and I have updated the tutorial [accordingly](https://github.com/peterhinch/micropython-async/blob/master/v3/docs/TUTORIAL.md). I do not know Russian but if you have a means of contacting the author you might want to point this out.

IhorNehrutsa, 2020-09-09:
I posted the info to the translator.
Thanks again for the job.

closed 2020-09-10 by peterhinch