peterhinch/micropython-async #48: AttributeError: 'module' object has no attribute 'Event'

oscar314, 2020-11-19:
Hi Peter,
I am newbie. I am trying to use queue.py from primitives folder but I dont understand where to find "Event" class/function in asyncio.
In query.py appears this line and triggers the error <<AttributeError: 'module' object has no attribute 'Event'>>
self._evput = asyncio.Event()  # Triggered by put, tested by get

I use Pycom device and I downloaded https://github.com/micropython/micropython-lib to use uasyncio but I didn´t find it in the whole lib. Thank you!

peterhinch, 2020-11-19:
I have no experience of the Pycom fork. My code is for version 3.0 which is built in to the current release of official MicroPython.  

I suspect that Pycom are using an outdated version of **uasyncio**. To check this, issue the following at the REPL:
```python
import uasyncio
print(uasyncio.__version__)
```
Version 3 will print `(3, 0, 0)`. Old versions will probably throw an exception.

kevinkk525, 2020-11-19:
> I use Pycom device and I downloaded https://github.com/micropython/micropython-lib to use uasyncio but I didn´t find it in the whole lib. 

You installed the uasyncio V2, which does not include Event and is not compatible with Peter's latest modules as those as for uasyncio V3.

romeocontrol, 2020-11-20:
As usual Peter is absolutely correct.
I really appreciate his work and his Tutorials. Read them, not only once!
I have used the new version with an gps-project on Lopy.
It did partially work when imported as normal lib.
You can find the V3 here:
https://github.com/micropython/micropython/tree/master/extmod/uasyncio

Good luck and happy hours!
RT

oscar314, 2020-11-20:
You were right. I have no version 3. Thank you for the alternatives.
I also appreciate this work and tutorials. It is very useful!!

closed 2020-11-20 by oscar314