peterhinch/micropython-async #49: usyncio tutorial

kindmartin, 2020-12-05:
hi all here / peter, Im looking for a tutorial for this new lib at 1.13 usyncio. I know is not the one Peter created here but I think after some days trying to do something I think any starting point is welcome. I was lookin these ones but still dont got it.

https://docs.micropython.org/en/latest/library/uasyncio.html?highlight=sched
https://gpiocc.github.io/learn/micropython/esp/2020/06/13/martin-ku-asynchronous-programming-with-uasyncio-in-micropython.html

for the records I need to run in after wake up a routine issuen 3 task with different frecuencies each, (i.e. #1 each 3 secs, #2 each 60 secs, #3 each 10 min) in a forever loop each one needs a few seconds to complete.
something like:

import uasyncio

   
async def AlarmedStatus(period):
    print('sendSMS(status_msg)')
    await uasyncio.sleep(period)
    
async def AlarmedLocation(period):
    print('sendLocationMsg()')
    await uasyncio.sleep(period)
    
async def wakeUpBySensor():
    print('SetWifiAP()   # prendo el AP para desactivar si owner se conecta ')
    await uasyncio.sleep(1)
    print('2G_Serial(1)')  
    await uasyncio.sleep(1)
    print('2G_EN(1)     # prendo el radio')
    await uasyncio.sleep(1)
    print('GPS_EN(1)      # prendo el GPS')
    await uasyncio.sleep(1)
    print('2G_DataEN(1)  # conecto los datos')
    await uasyncio.sleep(1)
    
    event_loop = uasyncio.get_event_loop() # creo un loop en asyncio
    print('lanzo las tareas')
    periodStatus = 3
    periodLocation = 1
    event_loop.create_task(AlarmedStatus(periodStatus))
    event_loop.create_task(AlarmedLocation(periodLocation))
    event_loop.run_forever()

wakeUpBySensor()

romeocontrol, 2020-12-05:
Hallo kindmartin!
Asyncio is more complicated than users hope. 
I am a novice as well and reading, trying, reading the tutorials of Peter Hinch is a must.
I would suggest to use an IDE like VisualStudioCode. It helps a lot.
Thonny is fine, but the console is not very useful for testing your own success.
Check all you can find in the repository MicroPython-async

Good Luck!
RT

kindmartin, 2020-12-05:
thanks will try VSC next days,
At first asyncio seems to me like that old approach like blynktimers.py or SimpleTimers in C/arduino where we set callbacks and then we run a forever loop in the timer class, but is more complex seems to be powerfull. I still digging in the structures.
Any other with a good starting point link to read?

thanks in advance.

M

peterhinch, 2020-12-06:
Please could you raise these queries in [the forum](http://forum.micropython.org/index.php). GitHub is for tracking bugs, and it is doubtful whether anyone other than myself will see these posts. 

In the forum more people will see and respond to support queries, and more people will be able to benefit from the replies. I will respond there to any request for uasyncio support.

closed 2020-12-06 by peterhinch