peterhinch/micropython-async #51: v3/as_drivers/client_server: socket close/server close

IhorNehrutsa, 2020-12-17:
I use 2 ESP323 to test server and client.
MicroPython v1.13 on 2020-09-02

In
https://github.com/peterhinch/micropython-async/blob/master/v3/as_drivers/client_server/userver.py
````
...
    def close(self):
        print('Closing server')
        self.server.close()
        await self.server.wait_closed()
        print('Server closed')

server = Server()
try:
    asyncio.run(server.run())
except KeyboardInterrupt:
    print('Interrupted')  # This mechanism doesn't work on Unix build.
finally:
    print('finally:')
    server.close()
    _ = asyncio.new_event_loop()
```
There is server.close() in finaly,
but I don't see 'Closing server' and 'Server closed'
```
Received ['value', 17] from client 3
Received ['value', 18] from client 3
Received ['value', 19] from client 3
Received ['value', 20] from client 3
Interrupted
finally:
```
Are socket and server still opened?

peterhinch, 2020-12-17:
You are right, and I'm baffled by this behaviour. The following test case behaves as expected:
```python
import uasyncio as asyncio
class Foo:
    async def run(self):
        for _ in range(10):
            print('running')
            await asyncio.sleep(1)
        raise RuntimeError('Got here')

    def close(self):
        print('closing')

foo = Foo()
try:
    asyncio.run(foo.run())
except KeyboardInterrupt:
    print('Interrupted')
finally:
    foo.close()
    _ = asyncio.new_event_loop()
```
Interrupting, either with **ctrl-c** or its internal exception, calls the `close()` method. Could this be a **uasyncio** issue?

kevinkk525, 2020-12-17:
There is a rather (obvious) mistake in the code. The "close" method is synchronous but awaits "await self.server.wait_closed()"
Sadly these kinds of errors seem to silently fail instead of throwing an exception.

peterhinch, 2020-12-17:
Oh dear, I should have spotted that. I've pushed an update which I believe fixes this. @IhorNehrutsa - comments/testing welcome!

Re **heartbeat.py** I've changed this so that ESP32 no longer throws an exception. I'm not happy about toggling pin 2 when there is no guarantee what might be attached to it. I've laid out and commented the code to make it easy for anyone with an LED on that pin to enable it.

IhorNehrutsa, 2020-12-17:
```
import uasyncio as asyncio
import utime as time

class Foo:
    async def run(self):
        for _ in range(5):
            print('running')
            await asyncio.sleep(1)
        raise RuntimeError('Got here')

    async def close(self):
        print('closing')
        t1 = time.ticks_ms()
        asyncio.sleep(2)
        t2 = time.ticks_ms()
        asyncio.wait_for(asyncio.sleep(2), 10)
        t3 = time.ticks_ms()
        print('t2 - t1 =', time.ticks_diff(t2, t1), 'ms')
        print('t3 - t2 =', time.ticks_diff(t3, t2), 'ms')
        print('closed')

foo = Foo()
try:
    asyncio.run(foo.run())
except KeyboardInterrupt:
    print('Interrupted')
finally:
    asyncio.run(foo.close())
    _ = asyncio.new_event_loop()

```
I'm not sure.
t2 - t1 not must be 2000 ms?
t3 - t2 must be 2000 ms?
```
running
running
Interrupted
closing
t2 - t1 = 0 ms
t3 - t2 = 11 ms
closed
```

IhorNehrutsa, 2020-12-17:
Is 
```
asyncio.wait_for(asyncio.sleep(2), 1)
```
must raise TimeOutError?

peterhinch, 2020-12-17:
```python
asyncio.sleep(2)
```
should read
```python
await asyncio.sleep(2)
```
and
```python
asyncio.wait_for(asyncio.sleep(2), 10)
```
should read
```python
await asyncio.wait_for(asyncio.sleep(2), 10)
```
See [this section of my tutorial](https://github.com/peterhinch/micropython-async/blob/master/v3/docs/TUTORIAL.md#522-tasks-with-timeouts).

IhorNehrutsa, 2020-12-17:
Peter, thanks for clarifying.

[async_test.zip](https://github.com/peterhinch/micropython-async/files/5712260/async_test.zip)
Why code
```
        await asyncio.wait_for(asyncio.sleep(2), 10)   #  <--- line 19
```
**raises TypeError: coroutine expected**
```
running
closing
Traceback (most recent call last):
  File "<stdin>", line 33, in <module>
  File "uasyncio/core.py", line 1, in run
  File "uasyncio/core.py", line 1, in run_until_complete
  File "uasyncio/core.py", line 1, in run_until_complete
  File "<stdin>", line 19, in close
  File "uasyncio/funcs.py", line 1, in wait_for
  File "uasyncio/core.py", line 1, in _promote_to_task
  File "uasyncio/core.py", line 1, in create_task
TypeError: coroutine expected
```
But manual 
http://docs.micropython.org/en/latest/library/uasyncio.html#uasyncio.sleep 
tells
**uasyncio.sleep() - This is a coroutine.**

peterhinch, 2020-12-18:
I think you've found a flaw in the docs. But in reality why would you set a timeout on a **sleep()**?

Please could you look at [my tutorial](https://github.com/peterhinch/micropython-async/blob/master/v3/docs/TUTORIAL.md#application-of-uasyncio-to-hardware-interfaces). This would have answered a lot of these questions. In particular your use of timeouts is incorrect because error trapping is [mandatory](https://github.com/peterhinch/micropython-async/blob/master/v3/docs/TUTORIAL.md#522-tasks-with-timeouts) in the calling routine.

I appreciate your pointing out the issues with my code. However there is a limit to the amount of time I have available for covering points which are described in detail in the tutorial.

IhorNehrutsa, 2020-12-18:
v3/as_drivers/client_server tested Ok

closed 2020-12-18 by IhorNehrutsa

peterhinch, 2020-12-18:
Thank you for confirming my tests.