peterhinch/micropython-async #54: distribute on pypi?

khusmann, 2021-01-28:
Thank you for this project!

Have you thought about hosting this as a micropython- package on pypi? It'd be a lot more seamless than having users manually copying primitives. (But perhaps there's a reason for this that I'm missing?)

Thanks again!

peterhinch, 2021-01-29:
This project is mainly documentation with a collection of code samples. Most of the primitives mimic the CPython primitives and I would expect them to be replaced with official versions when Damien finds time. So I don't think the project is an ideal candidate for PyPi.

khusmann, 2021-01-29:
> Most of the primitives mimic the CPython primitives and I would expect them to be replaced with official versions when Damien finds time.

Ah, makes sense. Thanks for the reply!

closed 2021-01-29 by peterhinch