peterhinch/micropython-async #55: v3 tutorial:  Event example not CPython compatible

miketeachman, 2021-02-16:
I'm using this tutorial to get up-to-speed with the changes in uasyncio V3 in MicroPython.

I discovered one thing today that may or may not be of interest.  I think one of the tutorial goals is to stick close to CPython behaviour in the examples.  If that's right, you might be interested in this minor issue that I observed.

Running the Event example in CPython produces this result:
```
$ python original_event.py
Task exception was never retrieved
future: <Task finished name='Task-2' coro=<waiter() done, defined at original_event.py:5> exception=RuntimeError("Task <Task pending name='Task-2' coro=<waiter() running at original_event.py:7>> got Future <Future pending> attached to a different loop")>
Traceback (most recent call last):
  File "original_event.py", line 7, in waiter
    await event.wait()  # Pause here until event is set
  File "C:\Python38\lib\asyncio\locks.py", line 309, in wait
    await fut
RuntimeError: Task <Task pending name='Task-2' coro=<waiter() running at original_event.py:7>> got Future <Future pending> attached to a different loop
Waiting for event
Setting event
Event is set
```

Googling the error I found a seemingly related post:
[asyncio.Semaphore RuntimeError: Task got Future attached to a different loop
](https://stackoverflow.com/questions/55918048/asyncio-semaphore-runtimeerror-task-got-future-attached-to-a-different-loop)

I modified the Event example, applying the solution from the above post:
```
import asyncio
from asyncio import Event

async def waiter(event):
    print('Waiting for event')
    await event.wait()  # Pause here until event is set
    print('Waiter got event.')
    event.clear()  # Flag caller and enable re-use of the event

async def main():
    event = Event()
    asyncio.create_task(waiter(event))
    await asyncio.sleep(2)
    print('Setting event')
    event.set()
    await asyncio.sleep(1)
    # Caller can check if event has been cleared
    print('Event is {}'.format('set' if event.is_set() else 'clear'))

asyncio.run(main())
```

After modification, the example runs in CPython.
```
$ python test_event.py
Waiting for event
Setting event
Waiter got event.
Event is clear
```

kevinkk525, 2021-02-16:
Yeah this is the typical problem of creating asyncio objects outside of a running loop. Creating tasks outside a running loop throws an exception too. However, with futures it works like asyncio.ensure_future(), which is rather old and deprecated now. 

Since in micropython there is always only one loop and we don't have futures, everything defaults to one standard loop. Besides that, it's often beneficial to create objects and tasks before a main loop is running as this can help with RAM fragmentation (depending on the application). But of course, all that is not CPython compatible. 

In this example it makes sense to create Event() in main() instead of outside but generally you might create an object with an Event created in its constructor. I often even start tasks in the constructor. And you typically don't create that object only while main() is running already.
Or am I the only one programming like that? 

Anyway, it would be good to be CPython compatible in the tutorial and outline any differences in behaviour between Cpython and upython.

peterhinch, 2021-02-17:
I will update the example accordingly.

Maybe I should include some text explaining this issue. I too instantiate events in constructors which usually get called before `asyncio.run()`. It hadn't struck me that this is incompatible with CPython.

miketeachman, 2021-02-17:
> Maybe I should include some text explaining this issue.

Readers would benefit for sure. 

After completing the V3 tutorial I thought of some addition material you might consider to make it rock even more:

- expand the section [a typical firmware application](https://github.com/peterhinch/micropython-async/blob/master/v3/docs/TUTORIAL.md#224-a-typical-firmware-app) to explain the under-the-hood aspects of `This requires the coro passed to asyncio.run() to await a non-terminating coro` and `await my_class.run_forever()  # Non-terminating method`.  I think there are some non-obvious subtleties to keeping the main() loop running.  For example, I was at first baffled when I created a task in main(), where the created task runs in a loop, waiting on an event.  I figured that this looping task qualified as the `non terminating method`.  It didn't work how I expected and the main() loop exited.  

- show an example of the Observer design pattern, built using asyncio.  It's a pattern that appears useful for getting async callbacks from another entity, such as an IRQ, or other task.  I'm in the process of using that pattern to make a new asyncio example for the rotary encoder repo that I maintain.

Thanks for continuing to maintain and expand your uasyncio tutorial.  It's helping a lot of us to master this important design approach.

peterhinch, 2021-02-18:
> For example, I was at first baffled when I created a task in main(), where the created task runs in a loop, waiting on an event. I figured that this looping task qualified as the non terminating method. It didn't work how I expected and the main() loop exited.

This "subtlety" is actually [a bug](https://github.com/micropython/micropython/issues/5843). The code you describe sounds correct and the fault is in `uasyncio`. Hopefully @Damien will fix this as it regularly catches people out.

There is work going on regarding interfacing IRQ's. The very latest `uasyncio` has `ThreadSafeFlag`. This is in the official docs and I have partially documented in the tutorial. It allows 1:1 signalling from a hard o soft IRQ to a coroutine. I am currently adapting the `Message` primitive with the aim of extending this to1:N signalling.

I think a detailed discussion of the Observer design pattern is perhaps beyond the scope of the tutorial - it's already rather big - but primitives supporting it will definitely be included. If you have any concise code samples demonstrating this pattern by all means show me and I might reconsider.

peterhinch, 2021-02-18:
I have added [this note](https://github.com/peterhinch/micropython-async/blob/master/v3/docs/TUTORIAL.md#note-on-cpython-compatibility) to the tutorial. Comments are welcome.

The `Message` primitive is amended as described above.

miketeachman, 2021-02-19:
> This "subtlety" is actually [a bug](https://github.com/micropython/micropython/issues/5843)

Thanks for mentioning this.  In the future I'll search the issues when I stumble on something that I think should work.  That would have saved me a couple of hours of disbelief and frustration.  To me, this is a major and fundamental bug in uasyncio V3.  I posted an encouraging note on the issue.

closed 2021-02-19 by miketeachman