peterhinch/micropython-async #56: DRIVERS.md document correction

pghpete, 2021-02-28:
First a thank you.  Thanks for this fanatic code and documentation!

I was following[https://github.com/peterhinch/micropython-async/blob/master/v3/docs/DRIVERS.md]( https://github.com/peterhinch/micropython-async/blob/master/v3/docs/DRIVERS.md) and it states to install the primitives directory on your hardware. I have done that then moved on to "**2. Installation and usage**" section.  Doing so, the first 3 `from primatives.[...]` imports work fine but the last does not.

```
import os

print(os.listdir('.'))
['boot.py', 'net_connect.py', 'primitives', 'pushover.py', 'sync_main.py', 'wifi_creds.py']

print(os.listdir('primitives'))
['__init__.py', 'aadc.py', 'barrier.py', 'condition.py', 'delay_ms.py', 'message.py', 'pushbutton.py', 'queue.py', 'semaphore.py', 'switch.py', 'tests']

print(os.listdir('.'))
['boot.py', 'net_connect.py', 'primitives', 'pushover.py', 'sync_main.py', 'wifi_creds.py']

print(os.listdir('primitives'))
['__init__.py', 'aadc.py', 'barrier.py', 'condition.py', 'delay_ms.py', 'message.py', 'pushbutton.py', 'queue.py', 'semaphore.py', 'switch.py', 'tests']

from primitives.switch import Switch
from primitives.pushbutton import Pushbutton
from primitives.aadc import AADC
from primitives.irq_event import IRQ_EVENT
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
ImportError: no module named 'primitives.irq_event'
```

I searched wthin the primitives directory/tree and could not find a file that provides irq_event.

Am I over looking something?

pghpete, 2021-02-28:
I think I answered my own question as I missed this...

_"The test for the IRQ_EVENT class requires a **Pyboard** with pins X1 and X2 linked. It is run as follows:"_

I will have to see if I can hack around the pyboard limitations so the instructions/code work on generic ESP32 WROOM-32 boards.

closed 2021-02-28 by pghpete

peterhinch, 2021-03-01:
You are correct: the `IRQ_EVENT` class has been superseded by the official `ThreadSafeFlag`. In my most recent update I removed the class and test script and updated the tutorial. Unfortunately missed these references in the drivers doc. I have now pushed an update.