peterhinch/micropython-async #57: Async Telnet ?

jdtsmith, 2021-03-07:
I thought I'd try out the new v3 on a nightly master v1.14 to implement a simple non-blocking telnet server.  To allow the terminal to function while the server is running, I'm running it in its own thread.  I derive from `IOBase`, but have to bridge between the non-blocking, sync desires of `uos.dupterm` and the blocking (select-polling?) async reader/writer provided by `uascynio.start_server`.  Below is what I have so far.  It does connect (and reconnect!), and can see strings sent from the client, but it inevitably leads to a `_thread.Guru Meditation Error: Core  0 panic'ed (LoadProhibited). Exception was unhandled.`, even, after some time, without any traffic (though perhaps the telnet client may send some commands at intervals).

Naturally it's hard to debug since you can't print things to the terminal without entering infinite recursion.  Any advice/thoughts would be much appreciated!

```python
import uasyncio as asyncio
from uio import IOBase 
import uos

class TelnetServer(IOBase):
    TELNET_PORT = 23
    async def run(self):
        print("Starting Telnet...",end='')
        self.server = await asyncio.start_server(self.accept, '0.0.0.0', self.TELNET_PORT)
        print("done")
        while True:
            await asyncio.sleep(100)
        
    async def accept(self, reader, writer):
        peer = reader.get_extra_info('peername')
        print("New telnet connection: {}".format(peer))
        self.writer = writer
        self.buf=b''
        #uos.dupterm(None) # Clear any old connection
        uos.dupterm(self)
        try:
            while True:
                try:
                    res = await reader.readline()
                    self.handle(res)
                except asyncio.TimeoutError:
                    res = b''
                if res == b'':
                    raise OSError
        except OSError as e:
            pass
        finally: 
            await reader.wait_closed()
            print('Telnet client disconnected: {}'.format(peer))

    def handle(self,res):
        while True:             # strip telnet commands
            f=res.find(b'\xff');
            if f is -1: break
            res=res[0:f] + res[f+3:]
        res = res.replace(b'\r\n',b'\n')
        self.buf += res
        
    def readinto(self,b):
        read_len = min(len(self.buf),len(b))
        if read_len:
            b[0:read_len] = self.buf[0:read_len]
            self.buf = self.buf[read_len:]
            return read_len
        else:
            return None

    def write(self,data):
        asyncio.create_task(self._write(data)) # fire and forget
        return(len(data)+2) # assume the best
        
    async def _write(self,data):
        self.writer.write(data+b'\r\n')
        await self.writer.drain()

import _thread
tns = TelnetServer()
def loop_in_thread(loop):
    loop.run_until_complete(tns.run())

loop = asyncio.get_event_loop()
_thread.start_new_thread(loop_in_thread,(loop,))
```

peterhinch, 2021-03-08:
The code looks good to me in its use of **uasyncio**. Unfortunately the application is outside of my experience, particularly with regard to threading on ESP32 which I have never used. Do you have other experience of using this? If not, it might be worth doing a forum search to see what its reputation is like. Have you checked the forum to see if anyone else has tried building a Telnet server?

I'm sorry I can't offer much help here.

jdtsmith, 2021-03-08:
Thanks for your thoughts.  There is [utelnetserver](https://github.com/cpopp/MicroTelnetServer/blob/master/utelnet/utelnetserver.py), which manages without threads or async.  I suppose it uses the "inherent" async of `socket.socket` (which I haven't really found documented).  I've taken to writing files with output for debugging purposes, and write seems to work from MP -> client.  I also didn't know if there was a recommended use of the [ThreadSafeFlag](https://forum.micropython.org/viewtopic.php?t=9867) I should consider.

peterhinch, 2021-03-09:
In general I deplore pre-emptive scheduling for [these reasons](https://github.com/peterhinch/micropython-async/blob/master/v3/docs/TUTORIAL.md#85-why-cooperative-rather-than-pre-emptive). Hence my lack of MicroPython threading experience.

My own socket based libraries use uasyncio with nonblocking sockets. However I have no idea whether this approach can be applied to Telnet as I haven't studied the protocol. My socket-based libraries are [resilient MQTT](https://github.com/peterhinch/micropython-mqtt) and [micropython-IOT](https://github.com/peterhinch/micropython-iot).

kevinkk525, 2021-03-09:
>  I suppose it uses the "inherent" async of socket.socket (which I haven't really found documented)

Probably the same as the webrepl, using a callback for sockets. It's a upy extension for use-cases like this.

kevinkk525, 2021-03-09:
Check this line in webrepl: https://github.com/micropython/micropython/blob/59a129f22f096d992496111c498b3ea97e637115/extmod/webrepl/webrepl.py#L24
It sets a callback for when the socket receives a connection and the following line sets a callback when the socket receives a message:
https://github.com/micropython/micropython/blob/59a129f22f096d992496111c498b3ea97e637115/extmod/webrepl/webrepl.py#L49

jdtsmith, 2021-03-09:
Thanks.  `utelnetserver` also uses the socket callback approach, which they got from `WebREPL`.  @kevinkk525, I had indeed forgotten the need to `uos.dupterm_notify(None)` on espXXXX (despite having myself [commented](https://github.com/cpopp/MicroTelnetServer/issues/4) on the issues caused by its removal from `utelnetserver`! sigh).  

After a few fixes, including this one, I actually have `async_telnet` working (with various warts to be dealt with)!  My hope is to implement telnet's "advanced" (circa 198x anyway) `linemode` so the speed and flexibility of local editing can be taken advantage of rather than "character at a time".   Once I get it working well enough I'll post. 

@peterhinch I definitely agree threading is problematic for the reasons your link mentions, and have had good success using `uasyncio` for hardware + MQTT IoT control.  Here the thread may work without much fuss since it's just spawning a background task which ties `dupterm` to the Asyncio event loop, without any communication back to the main process required. A bonus is it will run on another core on ESP32. I did find a good use of `ThreadSafeFlag` as well.  

BTW, when I start a server like  `await asyncio.start_server(self.accept, '0.0.0.0', self.TELNET_PORT)` and get a reader stream, does `await reader.readline()` simply perform a polling or select-based loop on the underlying socket?

peterhinch, 2021-03-09:
My understanding of MicroPython threading is that there is only one platform where a spawned thread runs on a different core, and that is RP2. On the ESP32 the OS runs on one core and MicroPython on the other, with all threads sharing the GIL. 

On RP2 there can be either one or two threads only. If there are two, each runs a separate interpreter and they do not share a GIL.

`uasyncio` stream I/O uses [the select.poll mechanism](https://github.com/micropython/micropython/blob/59a129f22f096d992496111c498b3ea97e637115/extmod/uasyncio/core.py#L69).

jdtsmith, 2021-04-05:
I stopped working on this while I tried to sort out the slow speeds which dupterm provides over sockets, which is I believe due to the lack of buffering.  See [this issue](https://github.com/micropython/micropython/issues/7029).

closed 2021-04-05 by jdtsmith

gretel, 2022-11-22:
https://github.com/shawwwn/micropython-ainput/blob/master/uaiotelnet/__init__.py