peterhinch/micropython-async #59: A useful usecase: REPL on REPL

ser, 2021-05-08:
I have an interesting suggestion for a use case for this documentation I hope: REPL on slave.

A second micropython (slave) board is connected to one of UARTs of the primary board (master). Using REPL on master board, we should be able to control REPL on slave, for example:

```
>>> import pyb
>>> remote import pyb
```
where ```remote``` is a software written using uasyncio which sends data to slave and receives answers. 

I suppose such a use case would be really valuable to understand micropython-async and useful in real life.

peterhinch, 2021-05-08:
Certainly. It sounds an interesting application. If you plan to do this and to publish on GitHub I will happily link to it in the docs.

ser, 2022-01-19:
I think this app https://github.com/peterhinch/micropython-async/blob/master/v3/docs/SYNCOM.md is sufficiently fulfilling the initial request, so I'm closing the issue.

closed 2022-01-19 by ser