peterhinch/micropython-async #62: ImportError: no module named 'primitives'

Curt-is-Pi, 2021-05-31:
Hello,

I am having an issue using the primitives modules. I get the 'ImportError: no module named 'primitives'' message any time I try to import a primitive. I can import uasyncio and print it's version though, 3,0,0. Any thoughts on what I maybe doing wrong? Keep in mind that I am Very new to programming so its most likely very basic .

peterhinch, 2021-06-01:
Please see [the docs](https://github.com/peterhinch/micropython-async/blob/master/v3/docs/TUTORIAL.md#111-primitives) for installation instructions.

bulletmark, 2021-07-02:
I just hit this issue as well. Searched around for a while then noticed that instruction in the docs and later found this issue. Can I ask why uasyncio is included in the MicroPython release but this part of it is not?

peterhinch, 2021-07-02:
The reason is that `uasyncio` is an official part of MicroPython written by @dpgeorge. This repo is entirely unofficial and maintained by myself. A subset of the primitives aim to replicate those provided with CPython `asyncio`. I expect that, in due course, official versions of those classes will be written and I will retire my versions.

closed 2022-11-25 by peterhinch