peterhinch/micropython-async #63: AttributeError: 'Server' object has no attribute 'task'

dubaleeiro, 2021-06-06:
I am trying to run a asynchronous webserver using [microdot](https://github.com/miguelgrinberg/microdot) libraries `microdot.py` and `microdot_asyncio.py` and came across the following error:

`Traceback (most recent call last):
  File "<stdin>", line 40, in <module>
  File "uasyncio/core.py", line 1, in run
  File "uasyncio/core.py", line 1, in run_until_complete
  File "uasyncio/core.py", line 1, in run_until_complete
  File "<stdin>", line 37, in main
  File "microdot_asyncio.py", line 165, in start_server
  File "uasyncio/stream.py", line 1, in wait_closed
AttributeError: 'Server' object has no attribute 'task'`

Does it have to do with my asyncio version? I am running MicroPython v1.15 with the [microdot](https://github.com/miguelgrinberg/microdot) files frozen into a custom build. 
Thanks!

peterhinch, 2021-06-06:
Given that `microdot` claims to support MicroPython, perhaps you'd be better off raising an issue there. It's not a package I've used.

miguelgrinberg, 2021-06-06:
@peterhinch Yeah, we've been discussing this issue with @dubaleeiro earlier today and it was my theory that this was a bug in uasyncio.

I just spent some time debugging it and found it to be a race condition in the server creation code for uasyncio. You can't basically call `server.wait_closed()` too soon after calling `create_server()`. Sometimes you may get lucky and everything works, other times, `wait_closed()` tries to access attributes of the server object that haven't been created yet.

I have now added a workaround on my side. Sorry for the trouble!

closed 2021-06-06 by dubaleeiro

peterhinch, 2021-06-07:
If you have found a bug in `uasyncio` please create a simple test case and raise an issue against it on [GitHub MicroPython](https://github.com/micropython/micropython) where @dpgeorge will doubtless address it.

dpgeorge, 2021-06-07:
> You can't basically call `server.wait_closed()` too soon after calling `create_server()`. Sometimes you may get lucky and everything works, other times, `wait_closed()` tries to access attributes of the server object that haven't been created yet.

That does sound like a bug in `uasyncio`, which should probably be fixed.

miguelgrinberg, 2021-06-07:
@dpgeorge Sure. While I was debugging this yesterday I came up with a fix as well, so I have created a PR: https://github.com/micropython/micropython/pull/7365.