peterhinch/micropython-async #67: Link/url broken #2

beyonlo, 2021-09-23:
Hi Peter!

In the link https://github.com/peterhinch/micropython-async/blob/master/v3/docs/I2C.md has a "The file sock_nonblock.py illustrates.." where "sock_nonblock.py" is a hyperlink to https://github.com/peterhinch/micropython-async/blob/master/v3/docs/sock_nonblock.py where this link do not exists.

Thank you.

peterhinch, 2021-09-24:
Sorry, I can't find this. I've searched I2C.md for "sock_nonblock" and also for "illustrates" and drawn a blank, both on the web and in my original text. Please could you point me to the error.

beyonlo, 2021-09-24:
Sorry, I pasted the wrong link.

The correct link is https://github.com/peterhinch/micropython-async/blob/master/v3/docs/TUTORIAL.md
"The file sock_nonblock.py illustrates.." where "sock_nonblock.py" is a hyperlink to https://github.com/peterhinch/micropython-async/blob/master/v3/docs/sock_nonblock.py where this link do not exists.

peterhinch, 2021-09-24:
Thanks for the report.

This one isn't as easy to fix: the file [sock_nonblock.py](https://github.com/peterhinch/micropython-async/blob/master/v2/sock_nonblock.py) is in the V2 tree and I never ported it to V3. For the moment I'll remove the reference while I decide what to do.

peterhinch, 2022-11-25:
Closing. That code sample is out of date and I don't plan to port it.

closed 2022-11-25 by peterhinch