peterhinch/micropython-async #69: sleeping for debounce_ms only makes sense when the state actually has changed

ernstkl, 2022-01-01:
in the code of the Switch class the sleeping for debounce_ms is done in each and every iteration of the 'while True' loop:
https://github.com/peterhinch/micropython-async/blob/c10a5c214c65e3a2a8c26f6685818dd3a53cac0b/aswitch.py#L137
I think that is wrong
- debounce is only required when there actually was a state change, so the sleeping should be done within the if clause
- the current code will delay reaction to a state change by up do 50ms, because the loop body is only executed once every ca 50ms
- if you move the sleep into the if statement as I suggest, you will need an "await sleep(0)" in the while loop body. So that it returns control to the event loop. Otherwise the rest of the code would never run.

peterhinch, 2022-01-01:
I accept your comments, but this was a deliberate design decision.

The reason is that the unconditional delay reduces the load on the scheduler, giving other tasks more processing time. In target applications switches are operated by people; a 50ms delay is only noticeable in fast gameplay. A sleep time of zero won't guarantee fast response because of competition with other tasks. Even with your patch response may be delayed by >50ms in a typical application. In other words the patch would trade higher processor overhead for improved _typical_ response times without imroving the worst-case response.

If you need to guarantee a short worst-case delay (for example in the case of a limit switch), interrupts are the way to go.

closed 2022-01-01 by peterhinch

ernstkl, 2022-01-01:
Thank you for your quick reply, and for your explanation! I agree that it makes sense in the general case to not poll the switch state as fast as possible. 

I could not see that reasoning from the code though, as "debounce_ms" is implicitly used for two different purposes.

Maybe you can make this explicit in the code (see my updated code in #70 as a suggestion) so you do not have to repeat your explanation in the future :-)