peterhinch/micropython-async #71: error in RMC parsing

ser, 2022-01-08:
My device submits an empty course value when the device is stationary, for example:

```
['GNRMC', '051003.00', 'A', '100', 'N', '100', 'E', '0.124', '', '080122', '', '', 'A', 'V']
```

which fact invalidates current parser here:

https://github.com/peterhinch/micropython-async/blob/c10a5c214c65e3a2a8c26f6685818dd3a53cac0b/v3/as_drivers/as_GPS/as_GPS.py#L328

I do not know what is your favourite solution to that so I am not submitting a pull request, but it should be fixed in some way :)

peterhinch, 2022-01-08:
Thanks for that. The solution that comes to mind is
```python
course = float(gps_segments[8]) if gps_segments[8] else 0.0
```
If you agree with this, please could you test with your unit. Or do you have a better approach?

ser, 2022-01-08:
it's perfect, thank you!

peterhinch, 2022-01-08:
Thanks for the feedback. Change now pushed.

closed 2022-01-08 by peterhinch

romeocontrol, 2022-01-08:
Dear Peter Hinch,
you are fast … as usual.
But.
A vehicle, that does not move, has a heading, i.e. the direction its nose points to. To get from a compass.
It has no course, as the course is the direction a vehicle MOVES on. No Move, no course.

Greetings romeo tango

ser, 2022-01-08:
I've noticed it before but I do not see a problem with heading 0 if stationary. It really does not matter whilst parsing the file later.

romeocontrol, 2022-01-08:
Hallo,
heading and course are different!
A gps-unit can only calculate a course, when it is moving. No move no course.
And it can not calculate a heading.
It is correct, not to send a course, when not moving. 
Or if the speed os zero, then the course is not usable.
Greetings
romeotango

ser, 2022-01-08:
I think you convinced me, Romeo, I will propose more subtle solution in few days after testing it, then. Thanks :)

romeocontrol, 2022-01-08:
Thank you for the reply.
What do you use it for?
I looked for some use cases with GPS and SAPOS to minimise the errors to some cm.
But it is a bit to complicated for me to use this system.
It ist just for Germany, either.
romeotango

peterhinch, 2022-01-09:
To get heading you need a compass (e.g. an IMU with a magnetometer).

You might try to infer it from the direction of the last motion, but this wouldn't work for a platform like a boat which can rotate around a point. Even a car can change its heading at the last moment before becoming stationary.

ser, 2022-01-09:
We do not have any other data than content of GNRMC message. What I am working on currently is to have no value which would work with other parts of the code.