peterhinch/micropython-async #72: How some stops the AS_GPS run coro?

ser, 2022-01-19:
I want my pyboard set the time from GPS and quit all GPS-related activities.

It's what I have:
```
async def gnss_time():
    sreader = asyncio.StreamReader(GNSS_UART)
    gps_tim = as_tGPS.GPS_Timer(sreader, PPS_PIN, local_offset=0,
                             fix_cb=lambda *_: blue.toggle(),
                             pps_cb=lambda *_: green.toggle())
    print('Waiting for TIME signal.')
    await gps_tim.ready()  # Wait for GPS to get a signal
    await gps_tim.set_rtc()  # Set RTC from GPS
    gps_tim.close()
    sreader.close()
    await asyncio.sleep(3)
    print("We have RTC date set from GNSS!")
```
but I can notice that closing is not enough, as I can still can see as_GPS/as_GPS.py updates occasional errors later in the console. 

My question is: how should I close it properly?

peterhinch, 2022-01-19:
The `uasyncio` stream `close()` method [does nothing](https://github.com/micropython/micropython/blob/037b2c72a1d5b54a5508a58ab2044628a7a39fa4/extmod/uasyncio/stream.py#L22). The best approach is probably to de-init the UART. See [doc ref](https://docs.micropython.org/en/latest/library/machine.UART.html#machine.UART.deinit).

ser, 2022-01-20:
It's shutting the communication on UART indeed, but when I decide to turn UART on after some time, the GPS update coro is waking up and running again. How to permanently shut down the GPS update coro after I got precise time?

peterhinch, 2022-01-20:
Why do you want to re-enable the UART? Presumably it is still connected to the GPS, yet you said you wanted to "quit all GPS-related activities".

ser, 2022-01-20:
My scenario is: get precise time and set RTC, then swap to location-only polling. At the moment I managed to achieve it utilising two UARTs connecting to the same GNSS device (it has multiple ports) - one uart gets time and shuts up, later second uart is listening to NMEA only.

peterhinch, 2022-01-20:
I think you might be able to stop the oroutine by replacing [this line](https://github.com/peterhinch/micropython-async/blob/dae2062279706078a91e8fe7dccfb86e8e902da2/v3/as_drivers/as_GPS/as_GPS.py#L177) with
```python
            self.run_task = asyncio.create_task(self._run())
```
You should then be able to stop the task by issuing
```python
gps_tim.run_task.cancel()
```

peterhinch, 2022-01-20:
I'm still puzzled as to why you need to do this. Your `gps_tim` instance inherits from `AS_GPS`: if you leave the coro running you should therefore be able to query the instance with `latitude()`, `longitude()` etc. This should remain the case even after you've issued `gps_tim.close()` to stop the seconds interrupt.

ser, 2022-01-22:
Hi Peter, I analysed my concept again in light of your comments and I've just integrated setting time and updating location into one coro and I do not need to stop as_GPS now indeed. 

But your suggestion related to stopping via ```.close()``` was working correctly either, as I have tested it. 

Thanks a lot, I have learnt a lot in this conversation.

closed 2022-01-22 by ser