peterhinch/micropython-async #74: Delay_ms leak

stephanelsmith, 2022-01-24:
Hello Peter, first, thank you very much for this incredible library.  Asyncio on micropython has been game changer for embedded applications.

Our application is fairly complex, we use context managers to build up and tear down the application in a controlled way.  We are noticing we have a memory leak, and I believe I've narrowed it down to Delay_ms.  

Give it a try:

`
import sys
import machine
import gc
import uasyncio as asyncio
from primitives.delay_ms import Delay_ms

async def main():
    async def scoped():
        nop = Delay_ms()
        await asyncio.sleep(.25)
    while True:
        await scoped()
        gc.collect()
        print(gc.mem_free())

try:
    asyncio.run(main())
except KeyboardInterrupt:
    pass
except Exception as err:
    sys.print_exception(err)
finally:
    asyncio.new_event_loop()
    gc.collect()
`

In primitives/Delay_ms, last line of __init__, the task is created but never canceled.  If you simply comment out that line, and re-run the test above, you'll see there's no loss.

`    def __init__(self, func=None, args=(), duration=1000):
        self._func = func
        self._args = args
        self._durn = duration  # Default duration
        self._retn = None  # Return value of launched callable
        self._tend = None  # Stop time (absolute ms).
        self._busy = False
        self._trig = asyncio.ThreadSafeFlag()
        self._tout = asyncio.Event()  # Timeout event
        self.wait = self._tout.wait  # Allow: await wait_ms.wait()
        self._ttask = self._fake  # Timer task
        asyncio.create_task(self._run())    ############### TASK CREATED BUT NEVER KILLED
`

In my code, I've done the following, in __init__
`self._mtask = asyncio.create_task(self._run()) #Main task`

and created a new function:
`def deinit(self):
        self._mtask.cancel()`

New Test case:
`
import sys
import machine
import gc
import uasyncio as asyncio
from primitives.delay_ms import Delay_ms

async def main():
    async def scoped():
        try:
            nop = Delay_ms()
        finally:
            nop.deinit()
        await asyncio.sleep(.25)
    while True:
        await scoped()
        gc.collect()
        print(gc.mem_free())

try:
    asyncio.run(main())
except KeyboardInterrupt:
    pass
except Exception as err:
    sys.print_exception(err)
finally:
    asyncio.new_event_loop()
    gc.collect()
`

I'm not 100% clear on __del__ operation in micropython, if it's guaranteed, but I think being explicit with a deinit function seems to make sense as other micropython objects also deinit.

stephanelsmith, 2022-01-24:
Just noticing delay_ms is used in other places like pushbutton.  Possibly need to add deinit there as well.

peterhinch, 2022-01-25:
> we use context managers to build up and tear down the application in a controlled way.

The primitives assume an orthodox firmware application which, once started, runs forever. You will find usage of **Delay_ms** and possibly other instances of such coroutines. That said, adding a **deinit()** method is a worthwhile enhancement.

> I'm not 100% clear on del operation in micropython

My understanding is that it is not guaranteed. Even in CPython the only guarantee is that it will run, at the latest, when you quit Python. However don't take this as gospel, it may be wrong or out of date.

peterhinch, 2022-01-25:
On further thought, stripping this down to its bones we have:
```python
import machine
import uasyncio as asyncio

async def rats():
    while True:
        await asyncio.sleep(1)

async def main():
    async def scoped():
        asyncio.create_task(rats())
        await asyncio.sleep(.25)
    while True:
        await scoped()

try:
    asyncio.run(main())
finally:
    asyncio.new_event_loop()
```
This will consume RAM _ad nauseam_ until the script is cancelled. Then the creation of a new event loop should remove all references to the created tasks and the RAM should be recovered. This accords with my observations: I've never seen a memory leak that persists after creating a new event loop.

When a class instance goes out of scope, it seems that tasks created by that task remain referenced by the event loop and remain running. Using the following script I have confirmed that CPython 3.8 behaves in the same way. So it's not a usayncio problem.
```python
import asyncio

class Foo:
    def __init__(self):
        asyncio.create_task(self.rats())

    async def rats(self):
        while True:
            print('Rats')
            await asyncio.sleep(1)

async def main():
    async def scoped():
        foo = Foo()
        await asyncio.sleep(.25)
    while True:
        await scoped()

try:
    asyncio.run(main())
finally:
    asyncio.new_event_loop()
```
The conclusion seems to be that it's the programmer's responsibility to explicitly cancel any coroutines started by class constructors or otherwise. As I said above, I don't believe you can rely on `__del__` to do this automatically. This could have further consequences as `Delay_ms`, `Switch` and `Pushbutton` can themselves launch coroutines, depending on the application.

I've never encountered an instance of this in a normal firmware application, but evidently instantiating a `Delay_ms` in a loop is problematic. You might want to consider creating each `Delay_ms` instance outside of a looping construct.

I'd welcome your comments.

stephanelsmith, 2022-01-25:
Hello Peter, agreed with your assessment, this is definitely not an uasyncio problem.

In my application specifically, we have nested contexts for 'robust' wifi/cell/uart/'you name it' gateways which are built up in layers, a physical context -> socket context -> mqtt core context -> mqtt application context.  Errors and disconnects are expected.  Each context is enclosed in a while loop + try block, so depending on errors raised, we can restart at the proper layer.  In parallel, other tasks are taking measurements, etc...  So, in this specific setup, we cannot normally restart the event loop without blasting everything.

Delay_ms is fantastic construct for both keepalive and watchdog functionality.  It's used through out including at every context level.  On __aexit__, tasks need to be canceled.  I missed this one!

The advantage of "deinit" being present may be more about awareness.  As you mention, it's not a problem for many applications as a machine reset or new loop will blow the issue away, but in specific applications where we have deeply nested tasks, task accounting is vital.  With the function present and/or a note in the documentation, those developers will know what to do.

peterhinch, 2022-01-25:
OK, we're both singing off the same hymn sheet :) I'll close this again.

closed 2022-01-25 by peterhinch

peterhinch, 2022-01-27:
I have made a minor change. I was concerned that, after issuing `.deinit`, the timer was in a non-working state. Attempts to retrigger it would fail silently. Now, in this event, a `RuntimeError` is raised. This is tested in `tests/delay_test.py`. The tutorial now has a brief explanation of the [issue](https://github.com/peterhinch/micropython-async/blob/master/v3/docs/TUTORIAL.md#44-object-scope).

The following note is so that I have a record of the design criteria.

#### Design

I investigated a design which avoided a `._run()` task, starting and cancelling a timer task as required. This would have had the advantage that an inactive instance which went out of scope would leave no running task. However maintaining the ability to trigger in a hard ISR proved problematic. Doubtless the issues with `micropython.schedule` could be fixed, but the code was starting to look rather elaborate. Further, it didn't fix the problem in the general case, as a _running_ delay which went our of scope would still leave a running task. It would still be necessary to issue `.stop` to ensure clean termination.

I have therefore retained the existing design.

stephanelsmith, 2022-01-28:
Documentation update is great, I would have made sense of this.  Thank you again for your attention to details and taking up this improvement.