peterhinch/micropython-async #77: Documentation Improvement: Pushbutton Suppress argument

bai-yi-bai, 2022-02-01:
Hi,
This library is awesome. I couldn't believe how simple it was to create tasks and to use buttons. I was able to do in hours what took me days/weeks in Arduino. Also, kudos that there is documentation on Github! 

However, it took me a few tries to figure out how to implement multi-function push buttons (single, long, and double). I think as micropython grows in popularity, a lot more students, non-native speakers, and people who are not computer scientists will use this library. I wonder if the documentation could be expanded to offer more "goal-oriented how to's". The trouble I had was that I initially set up a press_func, and then wanted to add more functionality from there. I'm sure many people will take a similar approach. After creating more tasks which used different methods, I began testing. Then I saw my console log fill up with undesired behavior. I'm sure many others will also get frustrated to see that when they try to set up double or long presses, also get short presses. It took me a few several reads, and a lot of experimenting, to figure out how to modify my code to use the suppress function correctly. For me, the concept of "suppress" didn't really enter my mind. https://github.com/peterhinch/micropython-async/blob/master/v3/docs/DRIVERS.md#411-the-suppress-constructor-argument

This would probably make for a great tutorial in a book, an article on a blog, or a youtube video, but why not expand this technical documentation here to make it more accessible to new users? Now, I'm not sure if there is a style guide that this project adheres to, or if the maintainers are against including animated gifs of both the console and a fritzing-style breadboard pressing a button. But here's my suggestion for a code snippet:

#### (4.1.1.1) Multi-Function Push Buttons Implementation
This code snippet demonstrates how to use which code to interpret push button single (release), double (presses), and long presses together without undesired events getting triggered. Setting the "suppress=True" argument is critical to making these events cooperate.
```
from machine import Pin
import uasyncio
import Pushbutton

btn = Pin(<your chosen pin>, Pin.IN, Pin.PULL_UP, suppress=True) # Assumes a Raspberry Pi Pico
pb = Pushbutton(btn)

async def short_press():
    print("SHORT")
    await uasyncio.sleep_ms(1000)
    
async def double_press():
    print("DOUBLE")
    await uasyncio.sleep_ms(1000)

async def long_press():
    print("LONG")
    await uasyncio.sleep_ms(1000)

async def main():
    q = queue.Queue()
    short_press = pb.release_func(short_press)
    double_press = pb.double_func(double_press)
    long_press = pb.long_func(long_press)
    while True:
        await uasyncio.sleep_ms(1000)```

peterhinch, 2022-02-01:
Thank you for your positive comments.

It's perhaps worth pointing out that you instantiate a `Queue` without using it. Also this:
```python
btn = Pin(<your chosen pin>, Pin.IN, Pin.PULL_UP, suppress=True) # Assumes a Raspberry Pi Pico
pb = Pushbutton(btn)
```
should read:
```python
btn = Pin(<your chosen pin>, Pin.IN, Pin.PULL_UP) # Assumes a Raspberry Pi Pico
pb = Pushbutton(btn, suppress=True)
```
You're the first user to report difficulties with the docs; I'll consider adding an example. Note there is [a test](https://github.com/peterhinch/micropython-async/blob/master/v3/primitives/tests/switches.py).

closed 2022-02-01 by peterhinch

bai-yi-bai, 2022-02-02:
Hi Peter, you're welcome! 

Thank for you for quickly replying and for fixing my errors. Oh, you're right, I abandoned using the queue in my project code and had left it in. I got here by way of the [How to Use Asyncio in MicroPython (Raspberry Pi Pico) | Digi-Key Electronics](https://www.youtube.com/watch?v=5VLvmA__2v0&ab_channel=Digi-Key) YouTube video with presenter Shawn Hymel. [Getting Started with Asyncio in MicroPython (Raspberry Pi Pico)](https://www.digikey.com/en/maker/projects/getting-started-with-asyncio-in-micropython-raspberry-pi-pico/110b4243a2f544b6af60411a85f0437c)

The main reason I think better documentation is so important is because so many students never get beyond time.sleep(). Documentation and education are two things I think a lot about. I like this project's github-based documentation hosting approach since it consolidates code and docs in the same space, rather than taking the user to readthedocs or elsewhere which the owner may pull down without notice or could go out of date sometime in the future.

Are you open to allowing pull requests to add to the documentation to provide more tutorials/how to's?

peterhinch, 2022-02-02:
I am, but I want any docs and code samples to conform to the approach I use in the tutorial. In particular I expect code samples to be examples of the best, most efficient, way to use `uasyncio`. So your example would become:
```python
from machine import Pin
import uasyncio as asyncio
from primitives.pushbutton import Pushbutton  # Installed as per the docs - e.g. __init__.py.

btn = Pin(18, Pin.IN, Pin.PULL_UP)
pb = Pushbutton(btn, suppress=True)

async def main():
    short_press = pb.release_func(print, ("SHORT",))
    double_press = pb.double_func(print, ("DOUBLE",))
    long_press = pb.long_func(print, ("LONG",))
    while True:
        await asyncio.sleep(1)

asyncio.run(main())
```
This example uses hardware I/O and therefore can't be run on CPython, general code samples should be capable of running under CPython or MicroPython with this code construct:
```python
try:
    import uasyncio as asyncio
except ImportError:
    import asyncio
```

peterhinch, 2022-02-02:
I have added the above script [here](https://github.com/peterhinch/micropython-async/blob/master/v3/docs/DRIVERS.md#411-the-suppress-constructor-argument). Thanks for the suggestion.

bai-yi-bai, 2022-02-02:
Awesome Peter!
That's great, I fully agree with the changes and rationale. It's great to see my suggestion implemented. I was afraid I was making extra work for you and wasn't sure how quickly I'd get feedback or how it would be taken... so I didn't want to put in too much effort initially. Thank you for taking my improvements; it also means I don't have to go through making a pull request. 

I realize now in my initial post, I didn't offer a suggestion for a re-write, which is bad form as a peer-reviewer. I've spent some time on some rewrites. I tried to add some conjunctions to grab the reader's attention and provided some scenarios which a user may want to try to mimic. Also, I was very explicit by adding the sequence of events in a numbered list. This helped to reduce the amount of mental math the reader has to do and eliminates <code> from regular paragraph text. This should make it easier to parse for the groups I mentioned (non-native speakers and students who may not realize what's going on).

H2 4.1 Pushbutton class
The ```primitives.pushbutton``` module provides the ```Pushbutton``` class for use with simple mechanical, spring-loaded push buttons. This class is a generalisation of the ```Switch``` class which normally supports open or normally closed switches connected to ground or 3V3. To a human, pushing a button is seen as a single event, but to a micro-controller, this one event is seen as combination of a rise and fall of voltage, or vice-versa. A long button press adds the component of time and a double-click appears as multiple voltage changes. The asyncio's pushbutton class provides the logic required to enable these more complex user interactions by monitoring these events over time.

Instances of this class can run a ```callable``` on on press, release, double-click or long press events.

H3 4.1.1 The suppress constructor argument
By default, whenever a button is pressed, the ```press_func``` runs immediately. This minimal latency is ideal for applications such as games. However, this behavior is not ideal for tasks such as navigating an on-screen menu, or setting the time on a clock. Therefore, the pushbutton class provides the ability to suppress 'intermediate' events and reduce them down to one single event. Enabling this behavior requires use of the ```suppress = True``` argument. To illustrate why, consider how long button presses and double-clicks are interpreted using the default library behavior. 

A long press is seen as three events:
1. ```press_func```
2. ```long_func```
3. ```release_func```

Similarly, a double-click is seen as five events:
1. ```press_func```
2. ```release_func```
3. ```press_func```
4. ```release_func```
5. ```double_func```

<Then continue with the rest of the text as is>
There can be a need for a ...  

Also, I also recommend changing the 5 instances of "double click" to match the 23 instances of "double-click".

peterhinch, 2022-02-02:
> This minimal latency is ideal for applications such as games. However, this behavior is not ideal for tasks such as navigating an on-screen menu, or setting the time on a clock.

This doesn't make sense to me. It wrongly implies that low latency is a problem. In particular, if setting a clock or starting a stopwatch you want a minimum of latency. If an application only requires the `press_func` then there is no need to use `suppress`: using it will add latency while providing no benefit.

The purpose of `suppress` is to disambiguate the response when an application requires either or both of long-presses and double-clicks. You suppress the single click until you are sure that there isn't a double or long click event. The additional latency in responding to a click is an unwelcome but inevitable side effect, as I attempt to explain in the doc. It is important that users are aware of this trade-off.

It's not the easiest thing to explain, but if we're going to change the wording, we need to get that idea across.

bai-yi-bai, 2022-02-03:
Thanks Peter, this is becoming clearer now.

> This doesn't make sense to me. It wrongly implies that low latency is a problem. In particular, if setting a clock or starting a stopwatch you want a minimum of latency. If an application only [...]

I was hoping I could get by without modifying the sentence about latency, but yes, I see my error. I think the concept of suppression should be first introduced the top-level down (application/usage), and then provide the technical details of the implementation. This will better serve multiple audiences. Again, sometimes words are not the answer, diagrams are probably better to show the multiple scenarios of button presses and function calls.

> The purpose of `suppress` is to disambiguate the response when an application requires either or both of long-presses and double-clicks. You suppress the single click until you are sure that there isn't a double or long click event. 

This is a great introduction to the section. I suggest leading off with this.

> The additional latency in responding to a click is an unwelcome but inevitable side effect, as I attempt to explain in the doc. It is important that users are aware of this trade-off.

I didn't really consider this as a trade-off, especially where human interactions and human perception has trouble at these time scales. I think we're both coming from different perspectives, but I think in the minds of a coder, implementing long-press and double-click detection will inherently introduce latency due to simple fact that the complex interactions require software monitoring. However, this can be made clearer, first in the initial paragraph, and then later on. Notice that I focused my editing  on the first paragraph, I avoided editing the rest of the 'under-the-hood' behavior which is invisible to the user.

> 
> It's not the easiest thing to explain, but if we're going to change the wording, we need to get that idea across.

Feedback accepted. Any comments on the other content I offered? For example, the 4.1?

Here's another attempt:

### The suppress constructor argument
The purpose of the `suppress` argument is to disambiguate the response when an application requires either, or both, long-press and double-click events. By design, whenever a button is pressed, the `press_func `runs immediately. This minimal latency is ideal for applications such as games. The pushbutton class provides the ability to suppress 'intermediate' events and reduce them down to one single event. The `suppress` argument is useful for applications where long-press, single-press, and double-click events are desired, such as clocks, watches, or menu navigation. However, long-press and double-click detection introduces additional latency to ensure correct classification of events and is therefore not suitable for all applications. To illustrate the default library behavior, consider how long button presses and double-clicks are interpreted.

A long press is seen as three events:

1. `press_func`
2. `long_func`
3. `release_func`

Similarly, a double-click is seen as five events:

1. `press_func`
2. `release_func`
3. `press_func`
4. `release_func`
5. `double_func`

There can be a need for a `callable` which runs if a button is pressed, but only if a double-click or long-press function does not run. The `suppress` argument changes the behaviour of the `release_func` to fill that role. This has timing implications. The soonest that the absence of a long press can be detected is on button release. Absence of a double-click can only be detected when the double-click timer times out without a second press occurring.

Note: `suppress` affects the behaviour of the `release_func` only. Other callbacks including `press_func` behave normally.

If the `suppress = True` constructor argument is set, the `release_func` will be launched as follows:
 1. If `double_func` does not exist on rapid button release.
 2. If `double_func` exists, after the expiration of the double-click timer.
 3. If `long_func` exists and the press duration causes `long_func` to be launched, `release_func` will not be launched.
 4. If `double_func` exists and a double-click occurs, `release_func` will not be launched.

In the typical case where `long_func` and `double_func` are both defined, this ensures that only one of `long_func`, `double_func` and `release_func` run. In the case of a single short press, the `release_func` will be delayed until the expiry of the double-click timer (because until that time a second click might occur).

The following script may be used to demonstrate the effect of this argument. As written, it assumes a Pi Pico with a push button attached between GPIO 18 and Gnd, with the primitives installed. 

```python
from machine import Pin
import uasyncio as asyncio
from primitives.pushbutton import Pushbutton

btn = Pin(18, Pin.IN, Pin.PULL_UP)  # Adapt for your hardware
pb = Pushbutton(btn, suppress=True)

async def main():
    short_press = pb.release_func(print, ("SHORT",))
    double_press = pb.double_func(print, ("DOUBLE",))
    long_press = pb.long_func(print, ("LONG",))
    while True:
        await asyncio.sleep(1)

asyncio.run(main())
```

peterhinch, 2022-02-03:
OK, I'll go with that. I've pushed an update, with a few very minor tweaks to the wording and formatting.

Thank you for the contribution.

bai-yi-bai, 2022-02-04:
You're welcome!