peterhinch/micropython-async #79: Results for 100 coros on an ESP32 at standard clock rate

fivdi, 2022-02-07:
This issue describes an observation rather than a problem. While running the coroutine demo [rate.py](https://github.com/peterhinch/micropython-async/blob/master/v3/as_demos/rate.py) on an ESP32 with MicroPython V1.18, the output was as follows:
```
Testing 100 coros for 2secs
Testing 200 coros for 2secs
Testing 500 coros for 2secs
Testing 1000 coros for 2secs
Coros  100  Iterations/sec  3104  Duration 322us
Coros  200  Iterations/sec  3154  Duration 317us
Coros  500  Iterations/sec  2870  Duration 348us
Coros 1000  Iterations/sec  2754  Duration 363us
```
The result for 100 coros was 322μs. This is quit a bit faster than the 920μs mentioned in the comment in [rate.py](https://github.com/peterhinch/micropython-async/blob/master/v3/as_demos/rate.py#L13). I guess there have been a lot of improvements to MicroPython since the measurements shown in rate.py were taken.

Please feel free to close this issue without changing anything if you wish.

fivdi, 2022-02-07:
Here's the output when the rate.py is run on an ESP32-C3 with MicroPython V1.18:
```
Testing 100 coros for 2secs
Testing 200 coros for 2secs
Testing 500 coros for 2secs
Testing 1000 coros for 2secs
Coros  100  Iterations/sec  4567  Duration 218us
Coros  200  Iterations/sec  4631  Duration 215us
Coros  500  Iterations/sec  4111  Duration 243us
Coros 1000  Iterations/sec  3831  Duration 260us
```
And here's the output when the rate.py is run on an ESP32-S2 with MicroPython V1.18:
```
Testing 100 coros for 2secs
Testing 200 coros for 2secs
Testing 500 coros for 2secs
Testing 1000 coros for 2secs
Coros  100  Iterations/sec  1402  Duration 713us
Coros  200  Iterations/sec  1449  Duration 689us
Coros  500  Iterations/sec  1433  Duration 697us
Coros 1000  Iterations/sec  1525  Duration 655us
```

peterhinch, 2022-02-08:
There has been a lot of discussion here on GitHub about ESP32 benchmarks. I am not knowledgeable about the architecture but I gather there are some pretty complex issues which can cause benchmark results to vary radically with small changes to the layout of code in RAM.

The radical variation between the ESP32 variants is interesting.

That said, I'll put release build 1.18 on my ESP32 reference board, re-run the benchmark and update the doc accordingly.

peterhinch, 2022-02-08:
I can confirm your result. I've updated the code comments with a note about variability.

Thank you for the information.

closed 2022-02-08 by peterhinch