peterhinch/micropython-async #86: Blinking lights not executing correcty.

VladoPortos, 2022-08-23:
Hello,

I'm trying to figure something out, I need on push / push long and double click execute series of events.

Right now I have this:

```
from machine import Pin
import uasyncio as asyncio
from primitives import Pushbutton

vm_name = "\'Windows 11\'"
start_command = "virsh start " + vm_name
reboot_command = "virsh reboot " + vm_name
force_shutdown_command = "virsh destroy " + vm_name

# PIN13 is D7 on DFROBOT ESP32 Beetle V1.0
btn = Pin(13, Pin.IN, Pin.PULL_UP)
led = Pin(27, Pin.OUT)
pb = Pushbutton(btn, suppress=True)
led.on()

async def main():
    short_press = pb.release_func(print, (start_command,))
    double_press = pb.double_func(print, (reboot_command,))
    long_press = pb.long_func(print, (force_shutdown_command,))
    while True:
        await asyncio.sleep(1)

asyncio.run(main())
```

And this is working fine, But when ever I want for example the led to blink on double_press adding:

```
def blinky():
    print(reboot_command)
    led.on()
    await asyncio.sleep_ms(1000)
    led.off()
    await asyncio.sleep_ms(1000)
    led.on()
```
and then calling just double_press = pb.double_func(blinky)

It executes everything, prints out all messages... or error.. I tried so many combinations, but I can't figure it out. 

I need:
the LED to turn on, on single press
the LED to turn off, on long press
the LED to blink a couple of times, on double press. 

Please help :)

peterhinch, 2022-08-23:
This should read
```python
async def blinky():  # This is a coroutine
    print(reboot_command)
    led.on()
    await asyncio.sleep_ms(1000)
    led.off()
    await asyncio.sleep_ms(1000)
    led.on()
```

VladoPortos, 2022-08-23:
That seems to did it ;)

closed 2022-08-23 by VladoPortos