peterhinch/micropython-async #90: Newbie issue with Async

Comalco, 2022-10-24:
I have been trying to get an old code example working and have been unable to understand an error I get in Thonny.
I have uasyncio package loaded, and delay_ms.py, pushbutton.py and _init_.py saved in my ESP8266.
I am hoping someone might identify and explain the problem for me please?

Code is this:
```
from machine import Pin
import uasyncio as asyncio
from primitives.pushbutton import Pushbutton

btn = Pin(12, Pin.IN, Pin.PULL_UP)  # Adapt for your hardware
pb = Pushbutton(btn, suppress=True)

async def main():
    short_press = pb.release_func(print, ("SHORT",))
    double_press = pb.double_func(print, ("DOUBLE",))
    long_press = pb.long_func(print, ("LONG",))
    while True:
        await asyncio.sleep(1)

asyncio.run(main())
```
Error is this:
```
Traceback (most recent call last):
  File "<stdin>", line 3, in <module>
  File "primitives/pushbutton.py", line 8, in <module>
  File "_init_.py", line 60, in __getattr__
ImportError: can't perform relative import
```

peterhinch, 2022-10-29:
I think you haven't got the right directory structure on your target. There needs to be a subdirectory `primitives` with `__init__.py`, `delay_ms.py` and `pushbutton.py` in it.

Comalco, 2022-10-31:
Hi Peter,
I thought it was that, but I checked and I have a directory with those three files in it on the ESP8266.
If I didnt, I would have expected an error to be thrown earlier than where Thonny says it is occuring??

I know my hardware is OK as I can run other programs using this button assignment (and the button on that pin) OK.
Ideas of what else to check?

[image]

peterhinch, 2022-11-01:
I think you have uncovered a bug in Thonny. I don't use this: I use official `mpremote` and Dave Hylands' `rshell`. I followed this procedure on an ESP8266 running firmware 1.19.1, using `rshell`:
 1. Create subdirectory `primitives`.
 2. Copy the three files into it.
 3. Create a file `rats.py` in the ESP8266 root directory containing your script.
 4. Issue `import rats`.
It ran without error, responding to all three types of button press.

What I think is happening (but it's hard to be sure looking at your screen image) is this. Thonny is replacing the double underscore of `__init__.py` with single underscores, yielding `_init_.py`. If I'm right this is a serious bug and should be reported to Thonny developers. As you probably know, the name `__init__.py` is significant to Python.

To prove or disprove this you'll need to use a tool other than Thonny: I suggest [mpremote](http://docs.micropython.org/en/latest/reference/mpremote.html). The following command will list the contents of the directory on a connected ESP8266:
```
mpremote ls :/primitives
```
You should see:
```
[adminpete@capybara]: /mnt/qnap2/data/Projects/MicroPython/micropython-async/v3
$ mpremote ls :/primitives
ls :/primitives
        1606 __init__.py
        2611 delay_ms.py
        5603 pushbutton.py
```
If you see single underscores, Thonny has altered the filename.

Comalco, 2022-11-02:
I suspect I have not put the double underscores when I copied the files to the ESP.
I loaded them in Thonny first, and then Saved them to the EP, and in Thonny you have to name the file when you save it this way.
As a newbie I didnt know about the double underscore.  I havent used this file before in the learning code I have done thus far.
Many thanks,

Comalco, 2022-11-02:
So, so I created the issue with not having double underscores in the init filename.
I changed that and then got a different error;
[image]
It really didnt make sense to me so I downloaded all three files again from Github, copied them into the primitives directory, and hey presto, now it works.  No idea why I still got the second error message (as above), but it seems to be fixed now.
Thanks for all the assistance Peter.      I still have a lot to learn about Python !!
I have 50K of code in a project I have been working on.  Hopefully this long press routine works better than my own.  I use a 3 button membrane switch to navigate some menus on a 1602 LCD display, and I use short and long press detection to assert different functions in only 3 buttons.   Thanks again.

peterhinch, 2022-11-02:
To understand `__init__.py` look up "Python packages" in any tutorial, e.g. [this one](https://www.tutorialsteacher.com/python/python-package).

Glad it's working now and that Thonny appears blameless.

closed 2022-11-02 by peterhinch