peterhinch/micropython-async #92: TypeError: 'WaitAny' object isn't iterable

obdevel, 2022-11-07:
Trying to wait on multiple events, which are set by the two coroutines that are created. These coroutines work fine if created elsewhere, and I can wait on their events individually. I'm working from the example code in TUTORIAL.md.

```
    ...
    from events import WaitAny
    ...

    async def sequence_test_coro(self):
        evp = asyncio.Event()
        evs = asyncio.Event()
        
        tp = asyncio.create_task(self.pubsub_test_coro(evp))
        ts = asyncio.create_task(self.sensor_test_coro(evs))

        while True:
            evt = await WaitAny((evp, evs))

            if evt is evp:
                self.logger.log("pubsub is active")
                evp.clear()
            else:
                self.logger.log("sensor is active")
                evs.clear()
```
Error output is:

Task exception wasn't retrieved
future: <Task> coro= <generator object 'sequence_test_coro' at 2000a460>
Traceback (most recent call last):
  File "uasyncio/core.py", line 1, in run_until_complete
  File "module_asyncio.py", line 231, in sequence_test_coro
TypeError: 'WaitAny' object isn't iterable

obdevel, 2022-11-08:
Solved.

The example given in TUTORIAL.md is incorrect. The correct usage is shown in V3/doc/EVENTS.md. 
e.g.

```
evt = await WaitAny((evp, evs))  # IS INCORRECT
evt = await WaitAny((evp, evs)).wait()  # IS CORRECT
```

peterhinch, 2022-11-10:
Apologies, you are right. Tutorial now fixed.

closed 2022-11-10 by peterhinch