peterhinch/micropython-async #93: Streaming device drivers example no longer functioning

fivdi, 2022-11-24:
In section 6.4 of the tutorial titled [Writing streaming device drivers](https://github.com/peterhinch/micropython-async/blob/master/v3/docs/TUTORIAL.md#64-writing-streaming-device-drivers) there is an example introduced with the text "The following is a complete awaitable delay class".

On a Raspberry Pi Pico, this example runs as expected with MicroPython v1.19.1 (rp2-pico-20220618-v1.19.1.uf2). However, when run with the MicroPython nightly build from 23.11.2022 (rp2-pico-20221123-unstable-v1.19.1-713-g7fe7c55bb.uf2) the program no longer runs correctly and doesn't display any output.

I don't know if this is a problem with MicroPython itself or with the example.

peterhinch, 2022-11-25:
Thanks for the report. 

Evidently something in MP has changed as the demo has worked. I have pushed a fix: the problem was my `.read` method returned `None`. I guess this was a bit naughty. Now it returns a byte. It runs on targets other than the Unix build. I've added a note to that effect and will investigate further.

peterhinch, 2022-11-25:
The Unix build handles I/O in a different way from other ports, as described [here](https://github.com/micropython/micropython/issues/7965#issuecomment-960259481). I have now added a general disclaimer in the tutorial [here](https://github.com/peterhinch/micropython-async/blob/master/v3/docs/TUTORIAL.md#63-using-the-stream-mechanism).

closed 2022-11-25 by peterhinch

fivdi, 2022-11-25:
Thanks, that fixes it.