peterhinch/micropython-async #95: What is an 'event-like' object ?

obdevel, 2023-01-18:
What attribute(s) does a class require in order to be 'event-like' and waitable by WaitAny / WaitAll ? This clearly works well with asyncio.Event objects but I'd like to use it with other classes. I added a wait() method to my class but that doesn't seem to be sufficient. What else do I need to implement ? Thanks.

peterhinch, 2023-01-20:
I attempt to explain this [here](https://github.com/peterhinch/micropython-async/blob/master/v3/docs/EVENTS.md#3-device-driver-design) in the first paragraph.

The `.wait` method must be asynchronous and it must pause until some triggering event occurs. Here the word "event" is with a small "e". In other words `.wait` pauses until something happens. The object behaves like an `Event` without actually being an instance of that class.

obdevel, 2023-01-22:
Thank you. It works fine now that I have my head around the precise syntax.
```
--> x = await WaitAny((sn1, sn2, )).wait()
--> x is sn1
False
--> x is sn2
True
```
Where sn1 and sn2 are objects that have an asynchronous .wait() method, as you describe.

These classes do have an asyncio.Event() object but I wanted to see if I could hide their internal implementation.

I also now have a version with a timeout, using a simple timeout class that sleeps then sets an event. If it completes first then the call to WaitAny has timed out.

peterhinch, 2023-01-22:
Glad you've got it working :)

closed 2023-01-22 by peterhinch