peterhinch/micropython-async #96: Encoder gives TypeError on nighly bluids

OlafM2015, 2023-01-21:
Testing with the latest builds  [v1.19.1-831-g4f3780a15 (2023-01-20) .uf2] the encoder with the test [program](https://github.com/peterhinch/micropython-async/blob/master/v3/primitives/tests/encoder_stop.py) 

Gives an TypeError, not sure what is wrong (on the [v1.19.1 (2022-06-18)] it works fine?

MicroPython v1.19.1-831-g4f3780a15 on 2023-01-20; Raspberry Pi Pico with RP2040

Type "help()" for more information.
>>> %Run -c $EDITOR_CONTENT
Running encoder test. Press ctrl-c to teminate.
TypeError: 
TypeError: 

Can you advise? Thanks

OlafM2015, 2023-01-22:
As far as I can see, the error occurs at (in [enconder.py](https://github.com/peterhinch/micropython-async/blob/master/v3/primitives/encoder.py)
```py
await self._tsf.wait()
```
waiting on the ThreadSafeFlag(). Do not see why this occurs in Nightly builds?

peterhinch, 2023-01-22:
Thanks for the report.

Something has gone wrong in builds since 20th December last year. I'm having trouble creating a minimal repro, but when I do I'll raise an issue. In the mean time please use an earlier build.

peterhinch, 2023-01-22:
I have raised [this issue](https://github.com/micropython/micropython/issues/10547).

OlafM2015, 2023-01-22:
Excellent, thanks

peterhinch, 2023-01-23:
Closing this as the maintainers have fixed it.

closed 2023-01-23 by peterhinch

OlafM2015, 2023-01-23:
Excellent, Tested latest nightly build, works properly. Thanks