peterhinch/micropython-async #98: Delay_ms cancel and cancel again

stephanelsmith, 2023-01-25:
In Delay_ms, if canceled and cancled again, you get and error that you cannont cancel() "None".   

```
    def deinit(self):
        self.stop()
        self._mtask.cancel()
        self._mtask = None
```

Recommending:
- I see there's a _fake timer that I don't remember from before (didn't notice), could we assign _mtask = _fake instead of None?  Same idea as used for _ttask in terms of double stop().
- test _mtask before cancel, skip if None
- Raise an exception if calling deinit twice shoul be an intentional error.   

Similar to the issue raised in the micropython issues area regarding gather, getting into the weeds of reliability.  Would not prefer to make deinit "safe" as I would like to not have to wrap all calls like this in try/except blocks.

Appreciate your consideration!  Happy to put together a PR.

peterhinch, 2023-01-26:
This seems simplest:
```python
    def deinit(self):
        if self._mtask is not None:
            self.stop()
            self._mtask.cancel()
            self._mtask = None
```

stephanelsmith, 2023-01-26:
Agreed, I'll put a PR together tomorrow.  Much appreciated for taking a look.

peterhinch, 2023-01-29:
Fix now implemented. Thanks for pointing this out.

closed 2023-01-29 by peterhinch

stephanelsmith, 2023-01-31:
Ya beat me to it!  Just getting a chance tonight.  Much appreciated.  Thank you.