peterhinch/micropython-async #100: Scheduling

Beormund, 2023-03-18:
Hi Peter,

I've been using your chron/scheduler successfully for a while now. Today I came across a slight issue. For a schedule: hrs=[7, 17]; wday= range(0,5) which is 7AM and 5PM Mon-Fri; if you then run this on a Saturday after 7AM the next run time produces Mon, 17:00 instead of 07:00.

Thanks.

Beormund, 2023-03-18:
I did some more testing and now I can't replicate it. It showing the correct next run time as Monday 07:00. I guess the only thing that's changed is WHEN I started running the schedule.??

peterhinch, 2023-03-30:
This flew under my radar because it was closed. I will investigate.

peterhinch, 2023-03-31:
Thanks for the report - I can confirm the bug.

This is a use case that I hadn't considered: where a sequence is scheduled to take place at a future time. So far it's proving a knotty problem. I'll give it my best shot over the next few days, but it is possible that it may end up being documented rather than fixed.

If I can't find a reasonably "micro" fix, workrounds are possible. For example schedule a task for 14:00 which runs the payload, then schedules three repeats at 15 minute intervals.

peterhinch, 2023-04-01:
There is a better workround which is to wait until just before the sequence is to start, then start it. The problem is that it looks ahead from the current time. If it sees a sequence it looks for the first entry after the current time. This is correct if the application has already waited on the `cron` instance, but not on initialisation.

I am leaning towards documenting this as fixing it without creating further corner cases is hard. By contrast the workround is easy.

I have written a simple script which allows sequences to be simulated to prove that they work as expected: users adapt the `sim` routine to match their proposed sequences. The following demos your case, with an initial delay, and shows that the sequence works correctly on the first and subsequent days.
```py
# simulate.py Adapt this to simulate scheduled sequences

from time import localtime, mktime
from sched.cron import cron

days = ("Monday", "Tuesday", "Wednesday", "Thursday", "Friday", "Saturday", "Sunday")
tim = 0  # Global time in secs

def print_time(msg=""):
    yr, mo, md, h, m, s, wd = localtime(tim)[:7]
    print(f"{msg} {h:02d}:{m:02d}:{s:02d} on {days[wd]} {md:02d}/{mo:02d}/{yr:02d}")

def wait(cr):  # Simulate waiting on a cron instance
    global tim
    tim += 2  # Must always wait >=2s before calling cron again
    dt = cr(tim)
    hrs, m_s = divmod(dt + 2, 3600)  # For neat display add back the 2 secs
    mins, secs = divmod(m_s, 60)
    print(f"Wait {hrs}hrs {mins}mins {secs}s")
    tim += dt
    print_time("Time now:")

def set_time(y, month, mday, hrs, mins, secs):
    global tim
    tim = mktime((y, month, mday, hrs, mins, secs, 0, 0))
    print_time("Start at:")

# Adapt the following to emulate the proposed application
def sim(*args):
    set_time(*args)
    cg = cron(hrs = 13, mins = 59)  # Wait until 13:59 before starting sequence
    cn = cron(hrs = 14, mins = range(0, 60, 15))
    wait(cg)
    print()
    for _ in range(10):
        wait(cn)
        print("Run payload.\n")

sim(2023, 3, 29, 15, 20, 0)  # Start time: year, month, mday, hrs, mins, secs
```
Output (on a Pyboard) - payload runs at expected times:
```
Start at: 15:20:00 on Wednesday 29/03/2023
Wait 22hrs 39mins 0s
Time now: 13:59:00 on Thursday 30/03/2023

Wait 0hrs 1mins 0s
Time now: 14:00:00 on Thursday 30/03/2023
Run payload.

Wait 0hrs 15mins 0s
Time now: 14:15:00 on Thursday 30/03/2023
Run payload.

Wait 0hrs 15mins 0s
Time now: 14:30:00 on Thursday 30/03/2023
Run payload.

Wait 0hrs 15mins 0s
Time now: 14:45:00 on Thursday 30/03/2023
Run payload.

Wait 23hrs 15mins 0s
Time now: 14:00:00 on Friday 31/03/2023
Run payload.

Wait 0hrs 15mins 0s
Time now: 14:15:00 on Friday 31/03/2023
Run payload.

Wait 0hrs 15mins 0s
Time now: 14:30:00 on Friday 31/03/2023
Run payload.

Wait 0hrs 15mins 0s
Time now: 14:45:00 on Friday 31/03/2023
Run payload.

Wait 23hrs 15mins 0s
Time now: 14:00:00 on Saturday 01/04/2023
Run payload.

Wait 0hrs 15mins 0s
Time now: 14:15:00 on Saturday 01/04/2023
Run payload.
```
I will post an update in the week. I'll update the docs and post this script. I also plan to update `crontest.py` to make it more cross-platform.

Beormund, 2023-04-01:
Thanks for looking into this. Does it only affect minute sequences? I don't seem to have any issue for hour lists but haven't checked hours with ranges/steps. I built a web UI around your scheduling utility so will give some thought about how to prepend the required crons for sequences:

[image]

peterhinch, 2023-04-02:
It affects cases where a sequence is scheduled to start at a future time. I have pushed a draft release of [SCHEDULE.md](https://github.com/peterhinch/micropython-async/blob/master/v3/docs/SCHEDULE.md) which explains the issue and suggests a workround. Comments and suggestions are welcome.

Beormund, 2023-04-02:
> It affects cases where a sequence is scheduled to start at a future time. I have pushed a draft release of [SCHEDULE.md](https://github.com/peterhinch/micropython-async/blob/master/v3/docs/SCHEDULE.md) which explains the issue and suggests a workround. Comments and suggestions are welcome.

Mmm. If a user can create a schedule at any time, and you can't rely on the next run time to be accurate for a sequence that might run in the future, how do you determine the delay_start schedule?

Beormund, 2023-04-02:
I'll have a play with the code and see if reversing the processing order of time specifiers helps.

peterhinch, 2023-04-02:
In the case of asynchronous code think I've found a solution. This makes no changes to `cron.py` which is a big benefit, as it is very hard to avoid creating new corner cases. I spent some time last week in the attempt before abandoning the attempt.

The idea is to use the techniques in `simulate.py` to calculate the time of triggers occurring after 0:00 on the current day. The aim is to calculate a time to wait before starting the sequence.

The first trigger occurring before the current time of day is `t0`, the first occurring after the current time of day is `t1`. Note that there may be no `t0`. In most cases you wait until (say) 5 minutes before the first trigger, then start the sequence. There are some corner cases such as fast sequences where this wait time needs to be shortened but I think these are straightforward.

If I'm right this can be done by adapting `schedule` and be completely transparent to the user. I will start on this tomorrow.

peterhinch, 2023-04-03:
I have pushed an update which aims to fix the issue in asynchronous code. The API is unchanged except that where an iterable is passed to `secs`, successive values must be at least 10s apart (formerly 2s).

Comments welcome :)

Beormund, 2023-04-04:
> I have pushed an update which aims to fix the issue in asynchronous code. The API is unchanged except that where an iterable is passed to `secs`, successive values must be at least 10s apart (formerly 2s).
> 
> Comments welcome :)

I did some testing. It's an elegant solution that works great - thank you! I extracted out the first long_wait (based on midnight last night) into a separate function - get_initial_run(**kwargs) so I could call it from within my get_next_run_time() function.

closed 2023-04-04 by peterhinch