peterhinch/micropython-async #101: Pushbutton double and long press events

schauveau, 2023-04-02:
I have a push button and I am trying to detect short, double and long push.
I noticed that if I do a _double long_ push so a double with a long second one then both `double_func`  and `long_func` are executed. That behavior contradicts the documentation : 

> _In the typical case where `long_func` and `double_func` are both defined, this ensures that only one of `long_func`, `double_func` and `release_func` run_

I can reproduce it with the 'suppress' example  from https://github.com/peterhinch/micropython-async/blob/master/v3/docs/DRIVERS.md#411-the-suppress-constructor-argument 

My current workaround is to add a global variable to disable 'long_func'  when it happens after a`double_func` without a `press_func`.

```python
button_ignore_long=False
        
def cb_button_release():
    print("RELEASE")
    
def cb_button_long():
    if button_ignore_long:
        return
    print("LONG")

def cb_button_double():
    global button_ignore_long
    button_ignore_long = True
    print("DOUBLE")

def cb_button_press():
    global button_ignore_long
    button_ignore_long = False
```

I also tried `pushbutton._ld.stop()` in the double callback but that does not seem to work. Not sure why !!!

peterhinch, 2023-04-02:
That is remarkable. The first iteration of the `Pushbutton` class was published about 9 years ago and ran under a different scheduler. In all that time nobody reported doing that test - it certainly never occurred to me!

Thanks for the report. In the `suppress` case it certainly shouldn't happen and I will fix it.

peterhinch, 2023-04-05:
I have pushed a fix. In the case of a short click quickly followed by a long press it now issues the double callback only, regardless of the `suppress` arg.

peterhinch, 2023-04-06:
Closing this as I believe this is now fixed. Thanks again for the report.

closed 2023-04-06 by peterhinch