peterhinch/micropython-async #114: Course/compass direction - wild fluxuations.

anglerfish27, 2023-11-18:
I don't know if this is a bug in the code, or just because of my signal or what. 
This is the 2nd GPS Adafruit "Ultimate" GPS module I have purchase, the first was to use to build an NTP server, which has been rock solid in providing my computers with highly accurate time. At the time of that buiild I believe I used Circuitpython and the code that Adafruit had so it was mostly plug and play except for the NTP server portion. I did notice this same behavior before it was put into PROD.

I purchase another unit a few weeks ago. Same one, with the battery backup coin cell. 
I purchased the active antenna they sell as well. I have the antenna facing the sky from a window (glass barrier in the way) other than the glass no obstruction to the open sky above.

I used your code (THANK YOU!!!!!) to get this up and running on an Unexpected Maker ESPS3 pro running the latest nightly build of MP so today's Nov/18, or maybe that means its yesterdays? Anyways..

No problems, I'm getting all the data nicely printed out that I want. Date/time and coordinates are spot on. The only blip in the system is with the GPS course/Compass heading. Even though the ESP32 is stationary as is the antenna. I am getting wild fluctuations in values from time to time. As much as 70-90 degrees. This isn't instant. I will have one bearing for awhile and then notice it switches to another after a random amount of time and so on. Thoughts on why? I have no less than 15 satellites in use based on your code and 9 in view (which is confusing to me as a newbie, I would expect those to be close or match) and the satellites in use never displays more than 6. The satellites numbers 68,69 ect.. will change from time to time.But its always [6 sats] displayed.

Little quirks and mostly not important to my project as long as lat/long/date/time are good which they are. But I would like to understand this behavior. 

I have a pyboard 1.1. maybe I will hook it up to that and play around with it since I know you used that in the beginning. I managed to find one on ebay that came with the Adafruit aluminum case. Looking closely at it it sure looks like the real deal. I use it for playing around I've got a small oled and a bme sensor on it right now :). Please never stop supporting those boards!!

Cheers,
Anglerfish27

chrisalbertson, 2023-11-18:
You should expect wild variation of heading on a stationary GPS.  Every second or so the GPS compute its location and it is always randomly displaced for where it was in the last second.  So the direction beteen the two points is random.

As we know, a GPS computes its heading by looking backward to where it was.  If that location data is noisy the heading will be all over the place.

In use there are two solutions
1) If the speed is low, below some threshold, ignore the heading or
2) Take averages over a longer period of time.

GPS is really usless at very small speeds as the noise is greater then the change in the location from point to point.    This is why robots have odometry and IMU and use filters to fuze the data.

I also have a GPS antenna mounted on my roof with a perfect view of the sky and it also jumps around maybe 30 cm between points.    But if I take a 24 hour average the data is much better.

Finally a bit of philosophy. What is the “heading” of a fence post that is stuck to the ground?  Most people would say the question is meaningless.  So it is wrong to even ask the GPS its heading if you know the antenna is stationary.

peterhinch, 2023-11-19:
@chrisalbertson is correct. Closing this.

closed 2023-11-19 by peterhinch